z88dk/z88dk #986: [z88][bug?] Sprite glitches?

Fabrizio-Caruso, 2018-09-24:
Sprites are wrongly displayed on the z88 target.

I am following @suborb's suggestion to define a window in my initialization code:
```
		static struct window win; /* Window structure */ 	
	
		win.graph=1; 
		win.width=255; 
		win.number='4'; 
		window(win); 
```

With the window definition above, the sprites are sort of displayed but very wrongly.

I must be missing some other important details on how to use sprites with this target.

[image]

suborb, 2018-09-24:
I've replaced the self-modifying version of putsprite with the generic version that will work when in a ROM!

A quick test shows that it works as well as graphics in general working on the z88.

closed 2018-09-24 by suborb

Fabrizio-Caruso, 2018-09-29:
Yes, indeed it does work.

commit efe9ec82 (2018-09-24): Fix #986: Use the non SMC version of putsprite for +z88