z88dk/z88dk #989: ABC800 relocatable binaries - malloc() issue

hpax, 2018-09-27:
Hello,

I am the maintainer of abc80sim and the abcdisk package, the latter which includes a program called bin2bac which converts a binary plus a .reloc file (produced by the --reloc-info option to z80asm) to a natively-loadable executable (simulating a BASIC program) for the ABC800 series.

However, the BASIC execution environment in ABC800 loads the binary at an unspecified address near the *TOP* of memory, with the stack following below that image. This has a few implications:

1. For a .bss section at a fixed address to the executable to be supported; the binary has to be padded with zeroes in order for it to be loaded at a low enough address.  I just added such support to bin2bac.

2. malloc() allocating above the binary will not work.  Instead, memory can be requested from the BASIC heap by calling the ROM function ALLOZMEM at fixed address 0x6c:

Input: DE = requested memory size
Output: HL = pointer to memory
DE = size of allocated memory
F.C = 1 if the memory area returned is smaller than the requested size

On *some* ROM versions the memory is zeroed, although I'm not sure how to determine which versions.

Alternatively and equivalently, the address of the end of the BASIC heap can be found at address 0xff0a.  The heap grows toward the stack, so the end of the heap is "however close you are willing to get to register SP".

The BASIC heap is never freed until END or CHAIN is executed at the end of a program; executing STOP (rst 10h; defb 0) will reset neither the stack nor the heap, and the program can be resumed with the commando CON (there may be some setup needed, I don't know.)

This applies to the BASIC execution environment specifically, when loaded though other means, such as an ABC-DOS .abs file, this does not apply.  I believe these situations can be disambiguated by examining if SP is located above or below the execution address.  Even if SP is located above the execution address, it might be a good idea to locating the heap below especially if loaded at an address of 48K or above; in that case the heap address variable at 0xff0a *should* still be valid, or the address of the beginning of memory (minus any loaded device drivers) can be found at address 0xff0c.

*** The following is highly optional and probably not particularly desirable, even ***

For very large programs it is possible to expose RAM at addresses 0-32K on ABC802 and ABC806, but *not* ABC800C or ABC800M.  Doing so will clobber the RAM disk (MEM: on 802, RAM: on 806) as well as the graphics (but not text) frame buffer on ABC806.  To do so:

ld a,0x05
out (35),a
ld a,0xe8    ; or 0xea for 40-column text mode
out (35),a

This maps RAM over the whole area, inclusive the text framebuffer at 30-32K (0x7800-0x7fff).  This of course means that swapping back becomes necessary to output on the screen or accessing any ROM functions, and that swapping needs to be done from an address above the 32K mark:

ld a,0x05
out (35),a
ld a,0x68   ; or 0x6a for 40-column text mode
out (35),a

Memory address 0xff54 should contain the current line width.

aralbrec, 2018-09-28:
A lot to chew on there.  I'll just pick on one item for now - the relocation information generated by `--reloc-info` was found to be inadequate.  It becomes complicated when the relocation is allowed to be by anything but multiples of 256 bytes.  This relocation information is needed in another target I am working on now so it may be an idea to fix it to supply relocation data to adjust programs by multiples of 256 bytes.

hpax, 2018-09-28:
This would not be suitable for the ABC series, unfortunately.  I'm not entirely sure why it would be hard to have relocations that aren't 256-byte aligned, given that the linker must have some way to represent them in the first place (or it wouldn't be able to relocate.)

I'm assuming that the problem is that zcc can split a 16-bit address between two 8-bit words, so that the "in-place addend" scheme of relocation is not applicable for the high half?  (Funny enough, I ran into the same problem myself back in '85 or so.)

This same problem actually exists even on some modern targets.  For that reason, the very modern ELF binary format has two ways of representing relocations: REL, which puts the addend as part of the data in the traditional manner, and RELA, where all the relocation information including addend is kept separately.  The downside of course is that RELA is larger.

I have to admit that based on the .reloc format, I had assumed that the only relocations that one needed to worry about were 16-bit relocation, and so I wrote my tools with that assumption in mind.

In fact, I just finished porting to C and cleaning up all my ABC family executable conversion tools; they are now a single tool called bin2abc which is in the abcdisk package:

https://git.zytor.com/abc80/abcdisk.git

zx70, 2018-09-28:
hello hpax, I wonder whether there is serious reason for not using a simpler block copy to a known location at a lower address and jump to it.. the only difficult thing to do is to avoid the copy loop to be overwritten, but it could be moved just before the dst block

hpax, 2018-09-28:
That would be fine, *except* that device drivers are commonly loaded in low memory.  So there is no really stable address, at least which doesn't cut the address space in half.

It also breaks the ability to treat the executable as if it was a BASIC program that can be RUN multiple times, or even SAVEd (yes, that works with my tools.  It's a matter of memory layout, as we are discussing.)

On another, what does zcc do about IX/IY?  In both the ABC series machines, IX is used as a pointer to the current file structure for file I/O, and IY is used to point to the system variable block.  On '80, it looks like IX/IY are saved and restored on call to a user routine, but I have not found any evidence that the '800 does the same thing.  Now, IX shouldn't be live at that point, since there is no file I/O being done, but IY is permanently set up, and calling ROM routines without IY being set correctly is not guaranteed to work.

It would be pretty trivial to insert those pushes and pops to the '800 loader, though; the setting around ROM calls would have to be a library issue of some kind.  The good news is that the value for IY can be treated as a constant on both systems: although in theory the system area is relocatable, in practice that never, ever happens.

hpax, 2018-09-28:
IY is 0xFE16 on '80 and 0xFF00 on '800.

hpax, 2018-09-28:
I just verified that IY is *not* restored on ABC800.  I'll add it to the .bac loader, though.

hpax, 2018-09-28:
It looks to me that sccz80 currently doesn't use IY; sdcc has a --reserve-regs-iy option.  I'm not sure if the latter has any performance penalty; in fact, I'm not sure which compiler zcc +abc80/800 will use.

I looked at the current z88dk git code, and there seems to be a *lot* of problems with the abc80/800 libc code; in fact the abc80 will touch I/O ports in a way that looks intended for ABC800, and most likely will disable the keyboard on '80.

If you could let me know what the right thing to do with IY is (see above), I think I might be able to relatively easily do some code changes.  These systems have a full generic file handling system that can have drivers in either ROM or RAM, so it ought to be pretty simple to do some fancy stuff.

suborb, 2018-09-28:
The default compile uses sccz80. As you’ve spotted, we prefer to only use one index register in classic - many machines use one or the other of them. Generally we stick to ix and the use the assembler to swap if necessary (eg zx81, vg5k)

There’s not a huge amount of abc80 specific code, as far as I know it’s only really stdio and graphics that have been used. So, any new contributions are welcome - it would be really nice to get another platform with file io. 

I’ve not had a chance to properly look at your tools, but do they solve the maximum size of 8k problem that z88dk currently has on the abc80?

hpax, 2018-09-29:
Yes they do - the size limit should be something like 13K for a 16K machine and 28K for a 32K machine.  The same applies to ABC800, except in device driver mode, however, device drivers rarely occupy a huge chunk of the system memory, one can hope :)

The ROM OS provides a pretty comprehensive set of functions, including I/O buffering for block devices.  For character devices it is a bit more tricky, as the interface is line-oriented, especially for reading (it's bascially fgets(), but you risk losing data if you don't save the whole line, depending on the device.) One of the really nice aspects is that the device interface supports console I/O as well, using an always-open file descriptor 0.  Unfortunately the '80 doesn't support reading a single character from the keyboard without echo except through a console-specific interface (fixed in '800).

One of the super bizarre aspects of ABC is that the text mode file format on a block device is completely frakking insane. As such, it gets to be imperative to keep track of text mode versus binary mode files; one of the ways to do that is to use the character device I/O functions even for block devices, but that might require another level of buffering :(

Either which way, I'm trying to figure out if there is a way to make use of this existing buffering without adding the full weight of an stdio layer...

aralbrec, 2018-09-29:
> I have to admit that based on the .reloc format, I had assumed that the only relocations that one needed to worry about were 16-bit relocation, and so I wrote my tools with that assumption in mind.

Yes that's part of the inadequacy.  Code can reference the low byte or the high byte independently.  If they reference the high byte then they need to know the low byte to find out if a relocation causes a carry out of the low byte into the high byte.

So the first thought was to create two relocation tables - one for the low byte and one for the high byte with the high byte table also containing the corresponding original low byte value.

But that is still not enough because when there is a non-trivial memory map, many binaries can be output (for different banks, for independent memory regions) and each of those binaries can be independently relocated.  So now you need to know in one binary information about where symbols come from and whether the corresponding binary is being moved.  This case I think you have to give up on without a full linker.

But I think the first two paragraphs can be done to allow relocation of a single binary blob or a single binary out of a set of binaries (like the code section and not the bss/data).  Issues with self-modifying code was also brought up.

Symbos is an example of a z80 operating system with a relocatable binary that can be partitioned into multiple sections with independent relocations.  It's accomplished by keeping everything in one binary and demarking each independent section with pointers.  Then the relocation code can know where symbols come from to do the appropriate arithmetic.  But I think I found it too was inadequate with regards to the split low/high byte referencing.

There is a short bit of discussion here:
https://www.z88dk.org/wiki/doku.php?id=libnew:target_embedded#generating_runtime_relocatable_binaries

aralbrec, 2018-09-29:
With regards to IX/IY, sccz80 does not use the index registers but the library will.  The classic library (where the abc target lives) tries to use one index register only (IX) and can be built switching IX/IY so that it uses IY rather than IX.  There are a few places where both index registers are used but I think that's in target-specific code only.  But a good plan is just setting IY to whatever you need before the system calls.

zsdcc can be told to not use IY with the --reserve-regs-iy but IY will still be used when the stack frame is large (like 120 bytes) which in itself is rare and should be avoided in normal code anyway.  The reserve-regs-iy mode is actually the preferred compile because it leads to smaller and better code when used in combination with the aggressive peephole rules SO3.