z88dk/z88dk #996: (classic) Add NEC PC-8800 support

suborb, 2018-10-04:
Gathering information:

* [ ] ROMS
* [ ] Disc format
* [ ] Tape format

Aimed support (phase 1):

* [ ] Generic console
* [ ] Inkey keyboard driver
* [ ] Lores graphics

Later on:

* [ ] Hires graphics

zx70, 2018-10-09:
https://github.com/z88dk/techdocs/blob/master/targets/pc8801/pc88-bank0.asm

zx70, 2018-10-13:
gathered from the _INKEY$ function:

00005AA5:       CALL 35CEh		; CHGET - Get key from keyboard buffer.
00005AA8:       JR Z,+09h		; _NO_KEY_FOUND
(...)

The 'A' register holds the resulting ASCII value.

For the output RST 18h should work, character code in 'A'.

I still couldn't define a suitable position for the machine code program, the way the BASIC program position is defined is a bit obscure,  but I suppose it will be right after 32768.   The ROM and its pages are always in the first 32K.

zx70, 2018-10-13:
OK, BASTXT (aka TXTTAB), holding the BASIC program poition points to.. 1 !
The BASIC programs seem to have plenty of RAM available, while the ROM picks the data with a page  routines at 44A4h, and 44D5h.   I think that the easiest way to get a working prototype is to deal with the LOAD command and transfer a binary block at 32768.

barbeque, 2018-10-14:
Is there a way I can help with this? I'm unfamiliar with the project but it would be great to have a working toolchain. I have an NEC PC-8801mkII with an HxC and the built-in machine language monitor, but the documentation on the latter is sort of thin on the ground.

zx70, 2018-10-14:
we'll surely need to test the results on the real hardware  :)
I suppose you are able to load from a wav file ?

barbeque, 2018-10-14:
I would have to build an adapter. According to the back of the machine, the port marked "CMT" is a DIN cassette connector, so it's probably designed to use a 'smart' cassette player, like on a TRS-80 Model 100, with a "remote" pin in addition to the "in" and "out," for stuff like auto start or precise rewinding.

I'll look into this some more, it's probably not that difficult - sounds like the pinout is the same on the MSX/FM-7/PC-6001 so we might not even need the remote.

This might come in handy for making the WAV: http://kitahei88.dtiblog.com/blog-entry-82.html

I was going to also look into loading binaries over serial. I have an HxC floppy emulator, so I could - in theory - build a disk image on the USB stick that contains a binary, although it would also need us to figure out the N88-BASIC filesystem and diskette format. Probably easier to go with the tape.

Also, the encoding is probably Shift-JIS and not necessarily strict ASCII, if that helps.

zx70, 2018-10-17:
I discovered that we have plenty of memory configurations here, the autoboot disks seem to deal with the n80 basic.. did this monster have 2 different rom BASIC??  .. then there is the disk basic.. etc..

barbeque, 2018-10-17:
Yeah, there is a DIP-switch on the front that toggles the computer between "N-BASIC" (from the PC-8001) and "N-88 BASIC," (obviously, from the PC-8801) along with a bunch of other settings (screen modes, serial modes, etc.)

DIP switch guide for the various machines:
http://www.kiwi-us.com/~ohta/pc88/dipsw.htm

suborb, 2018-10-18:
Sorry @zx70 I raised this one and then started on #1008 however you're finding out some good information.

I wonder if, rather than targeting cassette we should just jump straight to a bootable disc. In that scenario I'm guessing the IPL is available somewhere - does that have any keyboard/screen functions?

OTOH, I've noticed a lot of time it's easier to just skip the IPL/BASIC and read the keyboard directly, likewise with the screen.

zx70, 2018-10-18:
Agreed, but I often find the I/O function entries in the ROM first.
A first almost readable skeleton of the ROMs disassembly is now ready.
As said there are different approaches, a game could be loaded by the disk IPL process, the LOAD commands in N88 or N80 basic mode, from within a disk BASIC or (also used sometimes) by first entering in the MONitor and loading the binary block from tape.
I found some valuable link for ASM development etc:
https://qiita.com/tags/pc-8801
http://bugfire2009.ojaru.jp/download.html
P80MAKE

barbeque, 2018-10-18:
It might also be possible to load over serial, with something like `BLOAD "SER:"` but I've yet to find any confirmation on this.

barbeque, 2018-10-19:
That would square with what I've seen. Some games boot with a mention of "PC-8001 BASIC" instead of the N88 for just a brief second before loading.

Others jump into "SUPER SYSTEM 88," which seems to be a very minimal bootloader, probably popular with pirates, but I didn't think to check if the "N88-BASIC" LED on the front is illuminated then. I'll check that this weekend.

zx70, 2018-10-22:
I'd confirm the convenience (ad least at this stage) of keeping the ORG address at $8000.
I'm using t88tool_ver114 to test the library results, the keyboard scanning code is very basic (no debouncing, no capitals) but highlights the correct port/key data assignment.   RST$18 should be good for outputting text on both N80 and N88 modes, so a minimal support for console is now ready.

zx70, 2018-10-23:
http://retropc.net/cisc/m88/dl/pbios00s.lzh   "Pseudo BIOS" sources.
This should include a better keyboard handler, etc..

zx70, 2018-10-25:
I'm trying to get the appmaks support for the t88 format.  I'm leaving behind the disk format at the moment.

zx70, 2018-10-29:
I moved the ORG address to 8A00, it was chosen peeping into the original software written in the eighties.   My original choice of $8000 was probably overlapping a work area used by the monitor.
To run an appmake create program, first start N88 BASIC, get to the prompt (answer to "files?") then:
MON
R
G8A00

zx70, 2018-11-02:
Ok, now a test on the real HW would be welcome.
You could move in {z88dk}/examples/console and build adv_a.c or tic.c
Anybody wiling to help, please type:
zcc +pc88 -create-app -lm  -Cz--audio tic.c

... a WAV file will be generated.  Boot your PC8801 in N88 or N80 mode and type:
MON
R      (or L if you are in N80 mode)
(play your tape or audio file)
G8A00

In case it works please repeat all the steps by adding -Cz--fast to the command line, then post your feedbacks.

zx70, 2018-11-06:
160x100 low resolution graphics is now ready, I still need to understand how to hide the cursor (probably talking directly to the CRTC) but it containf few useful hints for the console driver.
Contributions are welcome, the recent improvements Dom made on appmake with the disk formats and the extended console library are very good and I'm a bit shy..

barbeque, 2018-11-06:
In N88-BASIC, the `SCREEN` command has a parameter to enable only the graphics screen, only the text screen, or both at once. If the cursor is still active, it probably means the text screen is still being overlaid.

There are a bunch of little mode switches I haven't had a chance to Google Translate yet, since it's slow doing it with the camera (no OCR on this PDF): https://archive.org/details/NECPC8801mkIIMRN88BASICN88BASICREFERENCEMANUAL1986L/page/n291

zx70, 2018-11-06:
Yes, I noticed it is the same for the CLS command.
The fact is that the low resolution mode I'm referring to is based on the text page, by switching the font to a special pseudo-graphics mode.
This mode could still mixed with the real graphics mode, so when it will be supported it sould be possible to overlap a "pixelated" low resolution picture over an high resolution one.

suborb, 2018-11-06:
@zx70 Can you share with me the emulator and ROMs you're using. I'm trying to set up takeda but I'm obviously missing something since all I get is the shortcuts printed at the bottom of the screen.

zx70, 2018-11-06:
The Takeda's emulator (and at least yet another emulator) take a while before completing the boot process.  I think it is the disk timeout on such emulators taking longer for some odd reason, or perhaps it was the right timing..

suborb, 2018-11-06:
Thanks, I'm obviously just impatient and didn't wait for long enough.

I've added a barebones gencon. Given that we've only got 40 attributes per line I'm not too sure how to handle colour etc - maybe a resolution of 2 characters will work ok.

zx70, 2018-11-07:
That's perfect,  this one wasn't easy but we are at an average stage already :)
I think that part of the work we've done can be reused on the pc8001  ;)

tschak909, 2018-11-09:
I soooo can't wait for this. Would love to bring PLATOTerm up on this! :D 

-Thom

zx70, 2018-11-13:
[image]

This shows the current cursor issue.   
I've also added a couple of functions good for this minimalistic approach to hide the FN keys bar when in native mode and to detect the BREAK key (CTRL-C, etc.. as detected by the services in ROM, all this stuff should be tested on different ROM variants before deciding to use them at startup).  Not very much really but they help to have a first skeleton.

zx70, 2018-11-15:
hires graphics.  I should be able to get something working in monochrome (green only) soon.

zx70, 2018-11-15:
This is the idea, to reuse the ready made "wide graphics" codebase and keep the default video mode as it is set at startup (640x200, mixed text & graphics).

[image]

.. the ADV_A in ghraphics mode could be rather interesting  :)

zx70, 2018-11-16:
This is an example obtained by booting the PC88 with 40x20 DIP switch settings.
To hide the bottom FN help bar there are 2 ways: 
- put clg() after printing chr$(12)
- use pc88_clearkey()

[image]

I consider the new graphics library at a good stage, line and 'mono sprite' drawing work as expected as well as the area fill.  I spent some time on porting the flood-fill function from the TS2068 but it doesn't work as expected, so I'm keeping it aside.  The memory organization I'm using wastes a lot of space but permitted me to reach the goal in short times.  The memory at $C000 up to the text area could be safely used by malloc or for data segments as long as no graphics date is put on it.  Fixing "appmake +nec" it could be possible to load code beginning at $100 or so, but lots of extra work is still required (ROM interposers, appmake fixes, etc..).

zx70, 2018-11-21:
The current z88dk port works in monitor mode, which means that the rom is paged in in the first 32Kbytes of memory.   The graphics pages overlap a 16K page at $C000 (which was already partially used by system variables, text memory page etc), so I lowered SP at $BFFF, which leaves 16K for the user program.   It is obviously limited, considering the machine capacity.
I'm planning to move the ROM interposed on the stack, so the first 32K can be recovered.
The interrupt pointers should be already initialized properly, so the appmake's NEC option, once adapted, should be able to move the ORG position to $100 and wrap the program by adding a BASIC program stub.

I'm not sure whether they are valid on all the PC88 models, but I succesfully tested the following instructions to page the RAM IN and OUT:

; page in
ld a,$11
out($e2),a

; page out
xor a
out($e2),a

Those simple instructions can be already used in MONitor mode to reach the extra 32K page for extra data storage (e.g. the Campus Lisp could be easily adapted).

zx70, 2018-11-22:
[image]

I prepared a minimal set of functions to make the native console support more practical (hiding the KEY help bar, changing the current text output position, etc..).
They should work in all the text resolutions (40x20, 40x25, 80x20 or 80x25).

zx70, 2018-11-23:
[image]

Jiffy interrupt intercepted to get the basic clock()  function  :)

0cjs, 2019-11-17:
> I would have to build an adapter. According to the back of the machine, the port marked "CMT" is a DIN cassette connector, so it's probably designed to use a 'smart' cassette player, like on a TRS-80 Model 100, with a "remote" pin in addition to the "in" and "out"....
> 
> I'll look into this some more, it's probably not that difficult - sounds like the pinout is the same on the MSX/FM-7/PC-6001 so we might not even need the remote.

The pinout is very likely to be the same; most Japanese 8-bit computers [seem to have followed a common standard](https://retrocomputing.stackexchange.com/a/12255/7208), at least as far as the research on that RC.SE answer has gone.

In general, you never need the remote pins; they're used only to allow you to stop the motor while PLAY is held down. If you execute the `LOAD` command before pressing PLAY, and the `SAVE` command after pressing PLAY, the computer will never know the difference.

yosi2112, 2020-07-15:
Here is a **list of I/O ports related to RS232C.**

| port(HEX) | Description |
----|---- 
| 20H | 8251 USART DATA |
| 21H | 8251 USART CONTROL (Mode[W]/Command[W]/Status[R]) *1|
| 30H | SYSTEM CONTROL |
| 6FH | BAUD RATE SELECT (FH~) |

# Description of I/O port 30H
| 7 | 6 | 5 | 4 | 3 | 2 | 1 | 0 |
----|----|----|----|----|----|----|----|
--- | --- | BS2 | BS1 | MTON | CDS | ~COLOR | ~40 |

| bit | desc | value
----|----|----
| BS2,BS1 | USART CH CONTROL | 00=CMT,600bps 01=CMT,1200bps 10=RS232C,ASYNC 11=RS232C,SYNC | 
| MTON | MOTOR SWITCH | 0=OFF 1=ON | 
| CDS | CMT CARRIER  CONTROL | 0=1200Hz 1=2400Hz| 
| ~COLOR | COLOR MODE | 0=COLOR 1=B/W | 
| ~40 | WIDTH SELECT | 0=40 1=80 | 

# Description of I/O port 6FH
Value | Baud rate
----|----
0000 (0) | 
0000 (0) | 
0000 (0) | 
0000 (0) | 
0000 (0) | 
0000 (0) | 

*1 see 8251 data sheet→[upd8251a.pdf](https://github.com/z88dk/z88dk/files/4925906/upd8251a.pdf)

suborb, 2023-03-26:
Closing this one since a lot of work was done on the PC-88 a couple of years ago.

closed 2023-03-26 by suborb