z88dk/z88dk #1008: (classic) Add Super-80 support

suborb, 2018-10-16:
This one is about 4+ machines in one so should keep me busy for a while. 

### Super80

- Z80 @ 2MHz
- 16k, 32k or 48k RAM (0000-BFFF)
- 12k ROM (C000-EFFF)
- 3.5k RAM (F000-FDFF), comes with the "64k ram" modification
- 0.5k RAM (FE00-FFFF) for Chipspeed colour board

16x32 character mapped at 0xbe00 by default

### Super80v

- Z80 @ 2MHz
- 16k, 32k or 48k RAM (0000-BFFF)
- 12k ROM (C000-EFFF)
- 2k Video RAM (F000-F7FF) banked with Colour RAM (modified Chipspeed board)
- 2k PCG RAM (F800-FFFF) banked with Character Generator ROM
-  MC6845 on $10 and $11

Information gathering:

- [x] Emulator is mame
- [x] Load via quickload
- [x] ROMs located

Probable features:

- [x] Mame quick load binary output
- [ ] Generic console
- [ ] Inkey keyboard driver

### Links

* https://forums.bannister.org/ubbthreads.php?ubb=showflat&Number=4031#Post4031
* http://avitech.com.au/?page_id=3088
* http://interbutt.com/mess/super80/

UncleBod, 2018-10-17:
The Electronics Australia magazine referred to in one of the [links](http://interbutt.com/mess/super80/ ) can be found [here](https://archive.org/details/EA1981/page/n23).

suborb, 2018-10-18:
I'm going to close this one now - the basics are there and any additional features can be new issues.

closed 2018-10-18 by suborb