z88dk/z88dk #1012: Limits on pre-2013 Visual Studio releases

zx70, 2018-11-09:
This is a low severity note,  I just noticed that with VS2010 it is not anymore possible to build appmake.
One of the problem is on the "designated initializers", (e.g. used to init cpm_discspec).
They are a C99 (or newer) feature, and older Visual Studio versions don't support C99.
Some feature appeared on VS2013 and only the later versions can be possibly used.

Taking apart the obvious alert for whoever is willing build z88dk with old compilers this means that the old VCPROJ files are definitely outdated an can be removed.

closed 2018-11-09 by zx70

aralbrec, 2018-11-11:
C11 is not well supported by Microsoft.  I think we should say min VS2015?

pauloscustodio, 2018-11-11:
I use VS2017 and it's support for both C11 and C++11 is good.

aralbrec, 2018-11-12:
I was under the impression it's a subset that happens to be in C++11.  Microsoft does not intend to develop the C compiler beyond C90 and only adds things that are also supported in C++, at least that's what they say in [public](https://github.com/MicrosoftDocs/cpp-docs/issues/407#issuecomment-417922008).

zx70, 2018-11-12:
I insisted to keep part of the old VCPROJ files as much as possible because I had the feeling that even few bits (appmake, etc..) could have been of some use even if built singularly.
Now almost all of them grew up and I must admit that it is just confusing.

Knowing the minimum usable VS version is useful, though..  we could point it out somewhere on the Wiki, but I wouldn't rely just on the official compatibility statements.
So far I understand that VS2017 is a working option, right ?

pauloscustodio, 2018-11-12:
I used VS2010 and VS2015 in the past and always found small incompatibilities with C11 and C++11 that annoyed me. Since I changed to VS2017, those issues ceased to exists.

I'm not claiming that VS2017 is fully-compliant to C11 and C++11, it just supports the set of functionalities I personally use from the standard.