z88dk/z88dk #1017: GFX: extending the generic color graphics support

zx70, 2018-11-20:
I'm using the PC88 as a reference target for color graphics.
The very tiny base already prepared for the EG2000 is being extended to more graphics functions.
The new function set will use the 'CALLEE' parameter passing trick.

* PLOT
* POINT
* DRAW
* DRAWR
* DRAWTO
* ...

suborb, 2018-11-20:
A few targets have colour graphics (through the backdoor so to speak) - (PC-6001, SPC-1000) They pick up the attribute from the console colours and then the regular b&w graphics routines can be used. 

This is quite nice because it ensures we've got some source-code portability and we don't have multiple sets of graphics routines - eg to plot in colour all you need to do is to set the ink() and just call draw()

Leveraging the gencon attribute setting code means that we can have both a uniform palette or custom palette with minimum hassle.

zx70, 2018-11-20:
This is an interesting point.   To have a global variable to store the current color is a requirement I found already (__gfx_color.asm).
Using the only "green" page on the PC88 makes the graphics run faster and permitted to quickly achieve the simple mono graphics support.   The way I chose to add colors seemed to me the pro one (clean parameter passing, recursion, CALLEE stuff, etc).
I'm still adding the 2 line functions I got already, then I keep this thread open for discussion  :)

zx70, 2018-11-20:
"Leveraging the gencon attribute setting code means that we can have both a uniform palette or custom palette with minimum hassle."  ..now I'm unsure  :)

zx70, 2018-11-20:
__CLIB_CONIO_NATIVE_COLOUR   <--   __gfx_color

suborb, 2018-11-20:
Yes, it's `-pragma-define:CLIB_CONIO_NATIVE_COLOUR=1` as the compile time option to disable the colour mapping.

Dealing with the machines with multiple planes and pixel perfect colour is awkward. The +rx78 was quite nice - it had different planes for paper (r,g,b) and ink (r,g,b) which made point() and screen$ easy to implement (i.e. look at the ink planes).

On the +lynx I used the alt-green screen to keep a monochrome version of the screen that achieved the same ends.

There's definitely one machine (multi8/lynx?) that allows writing to multiple planes at the same time which can speed up colour handling.

zx70, 2018-11-20:
I read about something similar (multiple planes at once)  also on the pc88 but I'm not still sure whether it is valid on all the models and what must be done to get it.   
isn't  __CLIB_CONIO_NATIVE_COLOUR the public variable holding the current colour code?

suborb, 2018-11-20:
The current colour is held on a port specific basis - it generally reflects the native format that is required (eg which nibble holds the ink/paper). Different screen modes sometimes need attributes as well so there's nothing shared across ports at that level. 

Take a look at the tiki100 graphics which support all 3 modes and the different attribute widths, or the MC6847 (SPC-1000, PC-6001) for a slightly more complex setup (blocks, monochrome + colour6 mode)

suborb, 2018-11-20:
From the takeda source it looks like port out (0x32),64 or out (0x33),64 (on PC-8001) and out (0x35),128 to toggle multiplane handling. There's no conditionals around different machines.

Once enabled there's this interesting bit of behaviour:

```
                        switch(Port35_GDM) {
                        case 0x00:
                                for(int i = 0; i < 3; i++) {
                                        switch((Port34_ALU >> i) & 0x11) {
                                        case 0x00:      // reset
                                                gvram[addr | (0x4000 * i)] &= ~data;
                                                break;
                                        case 0x01:      // set
                                                gvram[addr | (0x4000 * i)] |= data;
                                                break;
                                        case 0x10:      // reverse
                                                gvram[addr | (0x4000 * i)] ^= data;
                                                break;
                                        }
                                }
```

Where `i` controls the plane. `data` is what was written to the VRAM memory. It looks like there's "fast" copy capability as well. Using different GDM value the values for the 3 planes are kept internally and then written back. so ld a,(addr) ; ld (addr2),a doesn't actually use a.

zx70, 2018-11-21:
Very cool  :)
I reverted the last changes for the color drawing code, probably I should remove also cclg, cplot and cpoint (or replace them later with a macro).
If you don't mind I'd leave the evolution of the PC88 display and graphics to you, I'll dig on other aspects.

zx70, 2018-11-21:
Translating from here:
http://www.geocities.jp/submarine600/html/p8/vraminfo.html

"The extended access mode is a function to simultaneously access 3 planes using the ALU function newly established in SR and later models. "

suborb, 2018-11-21:
Ok, I'll pick the graphics now you've figured out all the hard stuff. I'm a bit short of time at the moment so it may take me a while, but I've started on graphical gencon - I never could get colour text mode to work which was a shame (the attributes always reflected B&W white)

It's a shame about the extended mode only being available on some models - I guess we could optionally use it. Do you know how to detect the models?

zx70, 2018-11-22:
> Do you know how to detect the models?
not yet, but it is part of the investigation I'm doing.

I think it depends on the "V2" graphics chip which was missing on the 2 earlier models, most of the existing games seem to have intentionally left them out to be reasonably fast.
I'm not sure but perhaps the extra CRTC commands are just ignored by the previous models.. if so perhaps it is possible to keep them running on a single color plane.

Since the PC8801 is a very flexible machine I had the strong temptation to add plenty of slightly different graphics libraries, but the risk is to increase the entropy uselessly, let's proceed by steps.

Lesson learnt, I'm now trying to sort out the existing functions for all the targets, changing the parameter passing to "CALLEE"..  I think I found a way to do it by steps still limiting the impact on the other architectures.   Extra functions may be eventually added once we have a neater base.

closed 2018-11-22 by zx70

zx70, 2018-11-22:
>>Do you know how to detect the models?
>not yet, but it is part of the investigation I'm doing.

A possible way is to trust the DIP switch settings:
in a,($31)
..bit 7 is reset for V2 mode

Here's a table listing the models and the available mode options
https://en.wikipedia.org/wiki/PC-8800_series