z88dk/z88dk #1020: Building z88dk on a Raspberry Pi fails

dhardingham, 2018-12-01:
When trying to build the latest nightly release, the following error is generated:

zcc +test -vn -a Issue_510_mult_minus1.c -o Issue_510_mult_minus1.opt
diff -w tmp.opt results/Issue_510_mult_minus1.opt
14a15,17
>       add     hl,sp
>       call    l_gint  ;
>       call    l_neg
Makefile:26: recipe for target 'Issue_510_mult_minus1.opt' failed
make[1]: *** [Issue_510_mult_minus1.opt] Error 1
make[1]: Leaving directory '/home/pi/z88dk/testsuite'
Makefile:175: recipe for target 'testsuite' failed
make: *** [testsuite] Error 2

I'm using a Raspberry Pi A+, running the latest version of Raspbian.

pi@raspberrypi:~/z88dk $ uname -a
Linux raspberrypi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

Have a found a new bug, or is there something wrong with Raspbian?

Regards
David

David Hardingham 

dhardingham@ntlworld.com

suborb, 2018-12-01:
That's definitely a problem!

I've managed to reproduce it (and another failure) on a pi I've got around the place so I'll investigate.

suborb, 2018-12-01:
I've fixed that one, but I think it means I should setup a pi/qemu setup for the nightly - annoyingly this issue isn't seen on a -m32 compile on Intel hardware.

dhardingham, 2018-12-02:
Many thanks.

From: suborb 
Sent: Saturday, December 1, 2018 11:33 PM
To: z88dk/z88dk 
Cc: dhardingham ; Author 
Subject: Re: [z88dk/z88dk] Building z88dk on a Raspberry Pi fails (#1020)

I've fixed that one, but I think it means I should setup a pi/qemu setup for the nightly - annoyingly this issue isn't seen on a -m32 compile on Intel hardware.

—

---
This email has been checked for viruses by AVG.
https://www.avg.com