z88dk/z88dk #1021: No warning on wrong CALLEE declarations

zx70, 2018-12-03:
This behavior happens only on very particular conditions, so it is not urgent, just nice to have.

Edit "string.h" and change the definition for "bzero_callee":

    //extern void __LIB__ bzero_callee(void *mem,size_t n) __smallc __z88dk_callee;
    extern void __LIB__ bzero_wrong_callee(void *mem,size_t n) __smallc __z88dk_callee;

Simple test program:

#include <string.h>
char *test[10];
void main ()
{
	bzero(test, 10);
}

When built with sdcc the proper error/warning messages are given.   sccz80 builds an half working program, (possibly with wrong stack usage), with no errors or warnings.

suborb, 2018-12-03:
If you compile with -Wimplicit-function-definition then you'll get a warning.

I guess it's disabled so that old code doesn't spam the console too much

zx70, 2018-12-03:
I see, thankyou  :)

suborb, 2018-12-03:
I might enable it by default and then fix up all the issues in the libraries and examples! Better to have diagnostics when we've got multiple calling conventions.