z88dk/z88dk #1030: LLVM Z80 backend

obiwanjacobi, 2018-12-18:
Just a heads-up.
Do you guys know about this: https://github.com/earl1k/llvm-z80
Maybe interesting to support to allow other (front-end) languages to generate z80 asm?
[2c]

suborb, 2018-12-18:
There was support added for this: http://www.colecovision.eu/llvm+sdcc/ which uses a C backend for llvm and then sdcc to compile the resulting code

aralbrec, 2018-12-22:
This is also integrated into z88dk.  You can build with llvm this way but one step still requires manual intervention to correct some things.  You stop at asm step, make corrections and continue to binary.

However the code is not as good as zsdcc by itself.  It might be interesting to compile other languages though and that was the idea when I tried it.

adriweb, 2022-04-02:
> The other various z80 back-ends for llvm out there hardly get started and then are usually abandoned. The one you point to hasn't changed in 5 years and it can't really do much now.

Maybe some of you know already though, but the modern/up-to-date one used by the TI CE calculators (eZ80 based platforms) community toolchain is available here: https://github.com/jacobly0/llvm-project/.
It also supports z80 and z180.