z88dk/z88dk #1034: [question] How to define the binary structure?

Fabrizio-Caruso, 2018-12-22:
Is it possible to define binary structure?

I mean somehing like what the CC65 linker cfg does.
Namely define where different code, data sections (with padding if necessary) go that for instance after loading everything is at the right place with no need to copy data to the location with extra code.

This could be useful for systems with video memory mapped onto RAM where graphics data has to be at specific locations. If the binary included code and dara with the right padding so that graphics data were at the right offset, then when you load your program you could get graphics data to be placed at the right locations with no code to move data. This could save some code at the cost of padding the binary sections.

aralbrec, 2018-12-22:
Yes the crt provides the memory map.  In normal compiles z88dk provides a default memory map but it can be changed by the user.

Fabrizio-Caruso, 2018-12-22:
@aralbrec Maybe the memory configuration could be factored-out and handled by a configuration file.
Is the crt code avoiding generating code that copies sections when the section are already at the right place?
Example: I generate a binary of 16k = exactly 8k of code (with padding) and 8k of graphics data.
On the target it is loaded at $0000 and so I have the the graphics data exactly at $2000, which is its final destination when the graphics chip of the target wants it to be.
Is the crt doing a useless copy of the graphics data?

aralbrec, 2018-12-22:
No the crt defines the memory map just like a linker script does.  The new library is designed for this so it allows the user to define a memory map and it will use that instead but the classic I think this is still harder, except I think you can define additional sections to add to the memory map easily.

The linker will produce binaries for each independent chunk with its own org.  It's appmake's job (or the user) to package this up in a suitable form.  Some targets have exe forms that define where things go in a large memory map whereas others don't.  These other ones appmake can sometimes build a loader that will load pieces into the right places.

Fabrizio-Caruso, 2018-12-22:
@aralbrec Maybe the "packaging" of the binary could be made simpler through a configuration file.
I though appmake received a merged raw binary as input in order to produce tape/disk/rom images.

CC65 lets the user define the sections and "packaging" of the raw binary thrugh a configuration file.
The user only changes the configuration file to change the structure of the binary. No need to code for the use for this.

obiwanjacobi, 2018-12-23:
The code identifies 'sections' in its source files and the memory-map gives order to those sections (code/data) which is all bound to a z88dk target. So currently I don't see an easy way to have a project-based config. You could try to override the memory-map itself, but that gets very technical fast.
It would be interesting how this configuration would work exactly. How do you refer to code and data in the right granularity?

Another use case that could be considered is (auto) configuring using some sort of rules. I'm hinting on producing segmented binary output for a system with bank-switched memory. Based on linker dependencies one could build a graph and auto-sort code chunks (functions) into piles of a certain size that are used together - although static dependencies are different from runtime call-stacks, it's the best we've got for now (I think). Of course these chunks should be relocatable/location independent.

aralbrec, 2018-12-24:
>     Maybe the "packaging" of the binary could be made simpler through a configuration file.
>     I though appmake received a merged raw binary as input in order to produce tape/disk/rom images.

There are hundreds of exe formats and many different memory architectures. A sensible memory map can be defaulted per target but the front-end tools would get very complicated fast if they had to deal with hundreds of output forms as well, not to mention maintaining that and adding new ones. Instead this job is moved to a standalone utility (appmake) which gets the output binaries, information in the map file and the name of the target with which it can put things together suitably. The intelligence is in appmake to build the suitable form; if something is missing appmake should be updated.

>     CC65 lets the user define the sections and "packaging" of the raw binary thrugh a configuration file.
>     The user only changes the configuration file to change the structure of the binary. No need to code for the use for this.

In z88dk the user doesn't have to do anything. He just says what he wants out - disk image, tape file, rom, intel hex and it happens automatically. The memory map is something else - this is something that can be legitimately desirable to customize. The default one provided by z88dk is sensible for the target for most situations but it can be the user has something else in mind. In the newlib, you indicate with a pragma that you want to provide your own memory map and it gets included in. Or you can append to the memory map and create new regions output as binary. Appmake can sometimes bundle these things properly depending on what is normal for the target machine.

aralbrec, 2018-12-24:
> The code identifies 'sections' in its source files and the memory-map gives order to those sections (code/data) which is all bound to a z88dk target. So currently I don't see an easy way to have a [...]

There are several things you can do now.

1. Make a completely custom memory map.  Usually you'd copy the library's and modify it to what you want.  This is sometimes done by users.  In the newlib, you indicate you have a custom memory map with a pragma which will cause z88dk to use the memory map stored in a specific file instead of its own.

2. You can append to the memory map.  This allows you to create new sections that will be output as independent binaries.

3. You can compile to consolidated object file or library and change section assignment with zobjcopy.  So yo can have control over what gets put together in the same binary(-ies) or what goes where if you are supplying a custom memory map.  It also gives the user control over where 3rd party library code gets placed in memory.

4.  The target may have special behaviour defined.  The zx next target, for example, has 2MB memory divided two ways as 16k banks or 8k banks (so two naming systems for the same memory) and an independent memory space in 8k chunks that can only appear at address 0x2000.  In this system, there are rules attached to the section names mapped to these memory regions.  Some naming rules enforce 16k or 8k boundaries and another allows code or data to flow to succeeding banks or pages.  Other naming rules allow you to create new section names that are inside the machine's banks.  So you can define section names inside BANK 40, for example, and even an ORG so that the placement inside the 16K bank is specified.  The linker will sequence stuff inside each bank appropriately.  The target also has exe forms that can describe what's in memory arbitrarily so appmake automatically makes exes that can cover the full 2MB space.  BTW for this target, it's found better if the programmer decides which banks things should be placed in rather than the link and I think that will turn out to be the case generally.  There is an exe type where the operating system loads into available memory banks instead of absolute ones but the programmer still allocates to banks, though they are logical rather than physical bank numbers.

5.  There is an appmake +glue target that generically defines multiple 64k memory banks and will build consolidated binaries for each bank.

As you can see the degree of complexity varies with target type but anything weird can be made bespoke inside appmake.

obiwanjacobi, 2018-12-24:
I have to take a look at this zx-next target - that's sort of what I am planning to do.
Thanx.

suborb, 2018-12-24:
I've just added support for `-pragma-define:MMAP=-1` to classic. This options allows you to define the complete section map in a local `"mmap.inc"` file. The only memory placement directive then in use is `CRT_ORG_CODE` which is for the start of the binary.

This means that you can use `ALIGN`, `ORG` as you wish in any section and have any layout you wish. You'll need to list all sections (see #860)

The downside is that you'll be completely on your own when it comes to generating a file that can be used on a real machine, but appmake may still be able to help with with `+glue`, `+inject`, `+extract`

Flowing around holes in the memory is covered by #832

zx70, 2018-12-26:
If I understood correctly,  one of the Fabrizio's concerns is to find a way to save memory in the same way he does with cc65:  "without any need to generate code that copies data at the wished address".
Is such code present in the classic library?   I thought it wasn't basing on a simple test where I noticed that the variables weren't initialized at the second run.