z88dk/z88dk #1036: [Question] How to compile C code to a library?

obiwanjacobi, 2018-12-23:
I have a custom target (zalt) and that has a basic asm core mainly for startup. I am developing C system code that should eventually (be optimized and) end up in the target but to keep the development cycle short I have that separate.

Problem is: I want to compile this C code into a lib in order to include that lib (and the target) in my test application projects.

I have tried to use the `-x` option on `zcc` but ik keeps generating a file called `a.bin` - probably some default when no explicit name can be found. I take it it does not see my `-x` option. 

Here's an example of a full command line I'm using:
`zcc +zalt -m -s --list -SO2 -DDEBUG -Ca-DDEBUG -clib=sdcc_iy -xzalt_lib @_zalt.lst`

What am I doing wrong here?

suborb, 2018-12-23:
The way it’s done in classic is to compile to .o files and then invoke z80asm with -x. 

I’ve never tried the direct zcc -x option to be honest.

obiwanjacobi, 2018-12-23:
Can you point to the source of that script?
I'm having trouble passing all the `.o` files to the `z80asm`.
I use a `.lst` file to list all the `.c` files, but now it seems I need to duplicate that with a `.lst` of `.o` files?

aralbrec, 2018-12-24:
[zobjcopy](https://github.com/z88dk/z88dk/wiki/Tool-zobjcopy) can manipulate libraries, for example it can change section assignmet or selectively expose symbols as public.

obiwanjacobi, 2018-12-24:
Yes, that's much better. Thanx.

So my mistake was to leave out the `-o` (output) option when using the `-x` (library) option, but you need both.

As for `zobjcopy`: I'm building from scratch - shouldn't need that.

closed 2018-12-24 by obiwanjacobi