z88dk/z88dk #1039: Trouble with z80asm -D, passing DEBUG flag

obiwanjacobi, 2018-12-26:
In my custom target startup 'asm' I have code like

```asm
	defs 0x0066 - ASMPC
	; address = 0x0066
	; !NMI
IF DEBUG
	call debug_monitor			; communicate debug info to SystemController
ENDIF
	retn
```

I have a target lib build script (like Winmake.bat) that does
`z80asm --IXIY -xzalt_sdcc_iy -D%1 -D__SDCC -D__SDCC_IY @target/zalt/library/zalt_sdcc_iy.lst`
the `%1` is passed to the build script and is `DEBUG` (checked that to make sure).

However in my listing file that I generate for a final binary I see:

```asm
                        	defs 0x0066 - ASMPC
100   0066              	; address = 0x0066
101   0066              	; !NMI
102   0066              IF DEBUG
103   0066              	call debug_monitor			; communicate debug info to SystemController
104   0066              ENDIF
105   0066  ED 45       	retn
```

This suggests to me that the `DEBUG` flag was not set.

What am I doing wrong here?

pauloscustodio, 2018-12-26:
Seams OK. Can you make a minimum-set of files to show the error and send?

aralbrec, 2018-12-28:
That code is in the crt which is only assembled when zcc is invoked to build a program in your build.cmd:

```
zcc +zalt -SO3 -DDEBUG -Ca-DDEBUG -clib=sdcc_iy @_zalt.lst -o zalt -m -s --list
```

If `-v` is added to the options I can see this near the end of the output:

```
PROCESSING CRT
z80asm -DDEBUG -D__SDCC_IY -I"c:\Projects\z80\z88dk\z88dk\lib\config\..\..\/libsrc/_DEVELOPMENT/target/z80" -D__SDCC --list   -s  -mz80 -s -I"c:\Projects\z80\z88dk\z88dk\lib\config\..\..\/lib" -I"c:\Projects\z80\z88dk\z88dk\lib\config\..\..\/libsrc/_DEVELOPMENT/target/z80"  "C:\Users\alvin\AppData\Local\Temp\zccA8A821.asm"
```

This is where the CRT is being assembled which includes your page zero code snippet.  The `-DDEBUG` option is there.

When you're building the library (the z80asm invoke in the original post) the crt isn't involved.  I'm not sure if something got mixed up here?

obiwanjacobi, 2018-12-28:
> The `-DDEBUG` option is there.

Yeah, but the code of the target (zalt.lib) has already been compiled, right? (BuildZaltLibs.cmd)
Or is a lib just bundled source code (instead of bundled `.o` files)?

So specifying DEBUG on zcc level will not magically re-assembly the target asm code... 

(The `--max-allocs-per-node` option was omitted on purpose in favor of faster compile times.)

aralbrec, 2018-12-28:
The library is not built with the crt in it.. the crt is only added to a compile when a binary is being generated.  

So the NMI asm, which is in the crt, will only be used with the zcc compile statement.  This is where DEBUG has to be defined for it to take effect.  The define is an asm-time define since the assembler must see "DEBUG" to decide what code is generated.  It's not something that can be built in the library - if it were and exported as "DEBUG" then the crt would have to EXTERN it and make a run-time decision (ie in code) to run the debug code or not.

obiwanjacobi, 2018-12-29:
> So the NMI asm, which is in the crt, will only be used with the zcc compile statement. This is where DEBUG has to be defined for it to take effect. The define is an asm-time define since the assembler must see "DEBUG" to decide what code is generated

So the -Ca-DDEBUG option I specify on the zcc should take care of that...

I fall into the trap of thinking building the zalt target library exposes all problems when in fact a full compile and link to a binary is required to have all code checked, compiled, assembled and linked.

I had typed an illegal asm instruction* and only saw that when I made a full binary build. Now that is something I would not expect. At least z80asm can check for valid instructions!?

*) I did a ld r, (nn), but had to load the address into hl and then do a ld r, (hl).

aralbrec, 2018-12-29:
> So the -Ca-DDEBUG option I specify on the zcc should take care of that...

Yes it should... did it work as expected?  If you can't get it to work I will try later today.

> I had typed an illegal asm instruction* and only saw that when I made a full binary build. Now that is something I would not expect. At least z80asm can check for valid instructions!?

Yes it should flag them when they are assembled.  Was it in the crt code so maybe didn't error until building the binary?  I think when building a library using `z80asm` it will stop rather than try to do all the asm files.  If it does all the asm files, the error message might be buried in the output text.  If there is an error, there should also be a *.err file amongst the source files.

obiwanjacobi, 2018-12-30:
I am unsure of how the crt/startup code is handled. `--list` options etc do not work on the `zalt_crt_0.asm` code. So that is a gray area for me. I can see the `.m4` is processed correctly and spits out the `,asm` and it is included in the final binary.

I will experiment with the `-Cl-DDEBUG` option, see if that fixes things...
`-DDEBUG -Ca-DDEBUG -Cl-DDEBUG`

aralbrec, 2018-12-30:
> I am unsure of how the crt/startup code is handled. --list options etc do not work on the zalt_crt_0.asm code.

The crt lis file should be named after the output file name.  So if you have `-o zalt` it should be named `zalt.lis`.  There are a lot of includes in the crt so you may have to scroll down or search for a familiar label like `Start` to get your bearings.  You'll have conditional assembly code in there too so to see what was actually put into the binary you can look at the assembled bytes column to see when a mnemonic or data was actually put in.

I think `-Ca` (assemble time) is probably the right one but it won't hurt to try `-Cl` (link time).

obiwanjacobi, 2018-12-31:
Yep, -Cl is the fix.
Just -Ca-DDEBUG result in:

```asm
668   0001                 ; exit program
669   0001              zalt_oblivian:
670   0001              IF DEBUG
671   0001                  halt
672   0001              ENDIF
673   0001  18 FE       	jr zalt_oblivian
```

Using the -Ca-DDEBUG and -Cl-DDEBUG options (together) results in:

```asm
668   0001                 ; exit program
669   0001              zalt_oblivian:
670   0001              IF DEBUG
671   0001  76              halt
672   0002              ENDIF
673   0002  18 FD       	jr zalt_oblivian
```

I stared at why I got continuous breakpoints (halt)-(and regs dumps, on my previously working programs until I checked all halt instructions in my code. Forgot all about this one ;-)