z88dk/z88dk #1041: sccz80 static vars have to be globally unique

obiwanjacobi, 2018-12-30:
I had a case where I had static vars of the same name in different source code files. Building the code under sccz80 gave an error that the var was already declared. Sdcc had no problem.

I did not expect this to be a problem.
Bug?

[2c]

suborb, 2018-12-30:
That's weird:

a.c:
```
static int a;

int main() {
}
```
b.c:
```
static int a;
```

`zcc +test a.c b.c` links without problems, as does putting the statics in functions. The one thing that doesn't work is having the same named static inside different scopes within functions.

Have you got an example of it failing?

obiwanjacobi, 2018-12-30:
I used sccz80 to compile and the new libraray if that matters.
I deleted the code but here are the historic links:
[Task2](https://github.com/obiwanjacobi/Zalt/blob/5ce3a1acaa86a38c6f351d74a05cc80a8c440e31/Source/z88dk_target/src/Task2.c) and [Task3](https://github.com/obiwanjacobi/Zalt/blob/5ce3a1acaa86a38c6f351d74a05cc80a8c440e31/Source/z88dk_target/src/Task3.c) both have _state and count static vars.

sdcc worked fine.

(I am in the process of converting the code base to be able to compile on both compilers, used sdcc previously)

aralbrec, 2018-12-31:
I can't reproduce any problems but there have been significant changes to sccz80 since the version you are using in 1.99B so it could be an issue here has been resolved.

closed 2019-01-02 by suborb