z88dk/z88dk #1043: z80asm silently accepts non-existent public label

obiwanjacobi, 2018-12-31:
If I declare a `public label` in and `.asm` file and refer to that from other files the final value/address is $0000.

I would expect that z80asm would check if the label that is declared as `public` is actually present in the `.asm` file. I took me a while to realize why my test program was continually resetting...

suborb, 2018-12-31:
I think there's some wrong with your install, this is a stripped down example which shows that the current code works as expected:

a.asm:
```
        PUBLIC  xyz
```

`z80asm a.asm`: Error at file 'a.asm' line 1: symbol 'xyz' not defined

b.asm:
```
EXTERN xyz
ld hl,xyz
```

`z80asm -b b.asm`:  Error at file 'b.asm' line 1: symbol 'xyz' not defined

obiwanjacobi, 2018-12-31:
It gets even stranger. This code is compiled into the zalt.lib (sdcc_iy) 

```asm
IF __SDCC
public _MemoryManager_SetPageAt
_MemoryManager_SetPageAt:
    ; two 8-bit parameters are pushed as one pair (16-bits) onto the stack
    ld hl, 2        ; to skip over return address
    add hl, sp      ; hl now points to pageIndex param
    ld b, (hl)      ; pageIndex in b
    inc hl          ; hl now points to pageId param
    ld d, (hl)      ; pageId in d
    ; b => page index, d => page-id
    jp  memorymanager_bank_page_write
ENDIF
```

I can see that the `__SDCC` is detected and code is being generated from the list file.

```asm
17    0000              IF __SDCC
18    0000              public _MemoryManager_SetPageAt
19    0000              _MemoryManager_SetPageAt:
20    0000                  ; two 8-bit parameters are pushed as one pair (16-bits) onto the stack
21    0000  21 02 00        ld hl, 2        ; to skip over return address
22    0003  39              add hl, sp      ; hl now points to pageIndex param
23    0004  46              ld b, (hl)      ; pageIndex in b
24    0005  23              inc hl          ; hl now points to pageId param
25    0006  56              ld d, (hl)      ; pageId in d
26    0007                  ; b => page index, d => page-id
27    0007  C3 00 00        jp  memorymanager_bank_page_write
```

I run the resulting library zalt.lib through z80nm and see this:

```
File zalt.lib at $124633: Z80RMF08
  Name: memory.thunk.SetPageAt
  Names:
    G A $0000 _MemoryManager_SetPageAt
  External names:
    U         memorymanager_bank_page_write
```

And my test program still resets.
Damn this is frustrating.

aralbrec, 2018-12-31:
> I can see that the `__SDCC` is detected and code is being generated from the list file.

```asm
17    0000              IF __SDCC
18    0000              public _MemoryManager_SetPageAt
19    0000              _MemoryManager_SetPageAt:
...
27    0007  C3 00 00        jp  memorymanager_bank_page_write
```

> I run the resulting library zalt.lib through z80nm and see this:

```
File zalt.lib at $124633: Z80RMF08
  Name: memory.thunk.SetPageAt
  Names:
    G A $0000 _MemoryManager_SetPageAt
  External names:
    U         memorymanager_bank_page_write
```

That looks right.  `_MemoryManager_SetPageAt` is marked `G` which means it's `PUBLIC` and `memorymanager_bank_page_write` is marked `U` for unresolved (I think) which means it's been declared `EXTERN`.  I don't see the `EXTERN` declaration in the code snippet but I assume it's there someplace.

Addresses are not resolved in lib files as the code hasn't been placed in memory yet by the linker.  So the address associated with a label will be an offset from a label value.

Addresses are also not resolved in .lis files as they are generated by the assembler before the link phase and they will also show addresses as an offset value.

You'll have to look at a map file generated by the binary build to know absolute values of labels.

aralbrec, 2018-12-31:
If the labels are still at unexpected values in the final binary (as indicated in the map file) then I'd like at the section your code is being assigned to.  If you haven't assigned code to a section it will likely take on a zero org.  In that case you should be seeing a non-empty `zalt.bin` or `zalt_.bin` file generated.  The output binaries are always named after a controlling section like `zalt_CODE.bin` and it it normally an error to have these other files generated without 0 size.

If the output is passed through appmake, they should at least generate a warning.

obiwanjacobi, 2018-12-31:
Hmm, I think I fixed it by adding a `section`.
I stopped doing that because I also fixed a crash a couple days ago by removing a `section`...

Thanks for the support, really appreciate it.

obiwanjacobi, 2018-12-31:
I have a custom memory model and I'm working toward compiling into separate pieces (modules) in order to load them into my memory using 4k pages. Thanx.