z88dk/z88dk #1044: How to convert an old target to the latest (nightly) version

obiwanjacobi, 2018-12-31:
Any guidance on how to get up to speed with the new target layout.
I remember there was a template (.zip) when I started, is there a new version for that?

I have installed the 31-12-2018 nightly.
Thanks.

obiwanjacobi, 2019-01-01:
Happy New Year!

I see zcc now also calls m4, nice!
I changed my config to use a `.m4` as crt startup file (like the z80 target).
But I get errors saying it can't find any of include(d) files.
I need to know what the base directory is for processing the m4 because I think it's trying to resolve the relative paths starting at the wrong location.

obiwanjacobi, 2019-01-01:
I copied the z80_crt.asm.m4 and it contains:

`include(`zcc_opt.def')`

This file is located in the _DEVELOPMENT root
but then also includes:

`include(`startup/zalt_crt_0.asm.m4')`

This file is clearly accessed from the target's root directory.
But the (root) path that is needed in order to get to these files is not the same!

Is this a bug in the z80 (and other) target(s)?
What does the command line look like to get zcc to compile any code for the z80 target?
(because the `z80.cfg` refers to the `z80_crt.asm.m4` this processing has to be done then)

feilipu, 2019-07-05:
There will always be an `*_UNASSIGNED.bin` generated.
This is only an issue if the file contains bytes. It should contain nothing.

If you turn on the `--list` option, I'm pretty sure you'll get an `*_UNASSIGNED.lst` file too, which should show you where the contents of the bin file come from.

feilipu, 2020-01-22:
There's some additional information on building a new target (here the [V-tech Genius Leader](https://www.z88dk.org/forum/viewtopic.php?id=10055)).

closed 2020-01-22 by feilipu