z88dk/z88dk #1046: sccz80 preprocessor differences to sdcc

obiwanjacobi, 2019-01-01:
Compiling for the sdcc I had created some of these preprocessor structures

```C
#ifdef __SDCC
StreamProvider *FastCall(ConsoleStreamProvider_Construct__fast(void *memory))
#elif __SCCZ80
StreamProvider __FASTCALL__ *ConsoleStreamProvider_Construct__fast(void *memory)
#else
StreamProvider *ConsoleStreamProvider_Construct(void *memory)
#endif
```
Now I'm compiling for the sccz80 and it has trouble preprocessing the files. My hunch is that `#elif` is not recognized. What would be a portable way to write this code in order to compile it on both flavors?

Also, sccz80 is not happy with this macro (with or without `{}`):

```C
#define dBreakpoint()               { #asm rst $30 #endasm }
```

suborb, 2019-01-01:
The #elif looks like it is parsed correctly (run with -E to view the intermediate output), though semicolons are missing at the end of the line if these are prototypes.

Both sdcc and sccz80 now support defining a fastcall function as follows:

```
StreamProvider *ConsoleStreamProvider_Construct__fast(void *memory) __z88dk_fastcall;
```

Re the macro, try this one for sccz80:  

```
#define dBreakpoint()  asm("rst $30")
```

and this one for sdcc (or sccz80 after the commit that will follow soon):

```
#define dBreakpoint()  __asm__("rst $30")
```

@aralbrec may have a neat intrinsic trick for doing the same thing without clobbering the sdcc optimisation.

obiwanjacobi, 2019-01-01:
Thanx!

Yeah I misinterpreted the error `unterminated // comment` - sorry. I thought that the if-endif matching was out of wack but it was literally a `//` comment that ended in `eof` instead of a new line.
Adding a new line to all my `.h` files fixed it...

suborb, 2019-01-01:
Cool, I'll close this one then.

closed 2019-01-01 by suborb