z88dk/z88dk #1050: sdcc: inline asm generates an error for rst

obiwanjacobi, 2019-01-02:
Compiling under sdcc(_iy) I encountered a problem with inline assembler for a RST $30 instruction - my implemenation of a breakpoint.

```C
void main(void)
{
    printf("Zalt says hello to Z88dk!");
    // dBreakpoint();
    __asm__("rst $30");
    //    __asm rst $30 __endasm;
}
```
Both flavors (one commented out) give the following error:
```
Error at file 'helloworld.c' line 570: integer '30' out of range
Errors in source file helloworld.c:
Error at file 'helloworld.c' line 570: integer '30' out of range
                   ^ ---- (null)
```
Funny thing is when I try a RST $00 it does not complain.

suborb, 2019-01-02:
I think some of the optimisation/transformation rules are stripping off the $, so try with 0x30 or the decimal value.

obiwanjacobi, 2019-01-02:
Argh, 30h works - it does say `integer` ...
I would expect the same syntax as z80asm...?

Anyway, thanks.

suborb, 2019-01-02:
It should be, but the rules transforming asxxx to z80asm syntax are dropping it somewhere. Since there's both regexes and patterns in those files it may be a case of a missing escape.

Anyway, at least there's a workaround.

aralbrec, 2019-01-25:
$ is a special character in asxxx assembly - it's used to indicate a local label.  Since local labels are not a feature in z80asm, the $ is stripped and a string is added to form a normal label with unique name.

The stripping is being done by a regular expression engine so it might be difficult to make it understand the difference between a label and a rst instruction.

aralbrec, 2019-01-25:
Just adding a note :- if compiling using zsdcc and using inlined asm, all inlined asm is treated as asxxx asm as it is embedded in zsdcc's native output.  This gets translated to z80asm syntax in post-processing so the inlined asm shouldn't really be regarded as z80asm.

asxxx syntax has some non-standard mnemonics.  For example index registers are accessed as in `ld l,5(ix)` instead of zilog form `ld l,(ix+5)`.  However, I think using standard zilog will pass through z88dk's translation and will be accepted anyway.

ped7g, 2019-01-25:
> The stripping is being done by a regular expression engine so it might be difficult to make it understand the difference between a label and a rst instruction.

If you put the problem like that, then you are completely correct, but if your modify the problem a bit, it may be less problematic.

You wrote about removing `$` from local labels, but label is identifier and it must start with letter/underscore/... (except if z80asm supports also the local anonymous labels like `2b`, but those don't need `$`). While `$` being used for hexadecimal `rst` value will be always followed by 0-9 digit. So I think modifying the stripping pattern to require non-digit as first char shouldn't break anything, but will fix `rst $..` syntax?

aralbrec, 2019-01-25:
The issue is user inlined code, not compiler-generated code as the compiler will never generate code where $ is used anywhere except at the end of label names.  User code is unstructured, contains comments, uses non-ascii character set, etc which poses a challenge for a regular expression engine trying to translate from a unique asm syntax to a standard one while also passing through correct asm syntax.

Although inlining a single instruction is not so bad, it's very hard to get people to stop inlining large swaths of code, like enclosing asm functions inside c wrappers.  These things should be in separate asm files with c header telling the compiler how to call them.   Inlined code damages the optimization and makes the c code non-portable, as well as raising issues with the regular expression translation step.

Anyway, I mention inlining user asm here because the fact users generate it means it's hard to ensure all cases of using $ is considered and taken care of.  The $ is not a hex prefix in asxxxx code (at least off the top of my head) and in any case numerical constants in asxxxx must be preceeded by a hash `#` so the `rst $30` instruction should actually be written as `rst #0x30` since as part of the c code, the asm is technically considered asxxxx.  But I agree with the OP in that you should be able to accept normal asm code here as well, or we should at least attempt to do that.

Some of the things a user might do with $:

```
label$:
jr label$
jr label$  ; comment
jr label$;comment
ld (label$),a
ld (label$30$),a
....
```

Looking for a number after $ would be difficult and error prone I think.  But what we can do is single out `rst $xx` in a special case before other $ rules are applied.

aralbrec, 2019-01-25:
@obiwanjacobi As mentioned inlined code prevents the peephole optimizer from optimizing code around the inlined asm.  The header "intrinsic.h" contains special intrinsics for inlining certain instructions that will not interfere with optimization.  So, for example, there is an `intrinsic_di()` function that will inline `di`.

There is no facility for inling `rst` there because `rst` is really a `call`.  So it is treated like that in z88dk; there is a way to inline rst and declare it as fastcall, callee with optional preserves_reg attribute by using function names beginning with `__RSTXX_`.  There is an example in the sega master system's library:

In the header file:

```
extern void SMS_crt0_RST18(unsigned int addr) __preserves_regs(b,c,d,e,h,l,iyl,iyh) __z88dk_fastcall;
extern void _RST18_SMS_crt0_RST18(unsigned int addr) __preserves_regs(b,c,d,e,h,l,iyl,iyh) __z88dk_fastcall;

#if __SMSLIB_ISRST_SMSCRT0RST18
   #define SMS_crt0_RST18  _RST18_SMS_crt0_RST18
#endif
```

The macro is giving the option to turn on RST substitution, otherwise the call will just be a regular one.  The rst substitution call uses C function name "_RST18_SMS_crt0_RST18".

The asm code must provide a labelled entry point at address 0x18 for the case where the user has chosen not to use rst substitution:

(in the page zero crt):
```
; address = 0x0018

   defs 0x0018 - ASMPC

   PUBLIC _SMS_crt0_RST18
   PUBLIC __RST18_SMS_crt0_RST18

_SMS_crt0_RST18:               ; Restart 18h - write HL to VDP Data Port
__RST18_SMS_crt0_RST18:
```

The rst substitution is done via peephole rule:

```
%"[ \t]*"0call%"[ \t]*"1__RST18%2
=
%0rst%1 0x18  ; call __RST18%2
```
where you can see a leading "__RSTXX" in the call name triggers a change to `RST XX`.

closed 2019-03-28 by suborb