z88dk/z88dk #1052: Implementing far memory

obiwanjacobi, 2019-01-04:
More than [two and a half years later](https://www.z88dk.org/forum/viewtopic.php?id=9351) I revisit the topic of far memory.

Now that I have basic software control of my MMU in place and working, I thought I would try my hand on implementing some sort of extended memory implementation.

The lp_* functions mentioned [here](https://github.com/z88dk/z88dk/wiki/FarMemory) (is this still up to date?) are what I am planning to implement. I think I have a reasonable understanding of how this is supposed to work, though there are some questions and I'm sure there will be more.

For working with data (structs, arrays strings) I can see how that is handled by the lp_x interface. There is even support for calling a mapped function. What I am unsure of is how one obtains the far pointer to begin with. I concluded so far that I will have to make a conversion method to make a far pointer from a regular one (I call it regular, because I also have a relative ptr - oh, great 3 ptr types! ;-). But to do that I will have to know where (in what 4k pages) things are located. Details on this are still in the dark for me...

Another question I have to answer is how I am going to load binaries into different banks. I think I will opt for generating multiple bin files using (memory map) `section`s and -if I remember correctly- `org`s. But this can wait until I have far data working (implies lots of new code already).

Of the `ehl` registers that contains the far ptr, I will use the `hl` pair as a regular cpu (64k) memory ptr. From that ptr I can deduce what 4k page the code (or data) was targeted to. That current 4k page will be swapped out and replaced with the one that is indicated in the `e` register. The `e` register can exactly contain (what I call) a PageId that uniquely identifies one of the 4k pages in the 1mb total address space.
I will have to check not to swap out the 4k page I am currently executing code on. Is there a way to signal an error from the lp_x functions? Set the carry flag? Ideally this type of error is detected at 'compile' (or link or bin-building) time...

[More info on my Memory Management system can be found here.](https://docs.google.com/document/d/1tF7p2I6861tk6l0KzyiMkYkhx_dNrJLCjcLoVhHxSHA/edit?usp=sharing)

Please let me know if I am doing this wrong, because I have noticed that the docs are not always up to date to the latest changes and your knowledge about this specific topic is what I am looking for.

suborb, 2019-01-07:
Nothing has changed with the far pointers in the past couple of years apart from the addition of far function pointers and the trampolined SHORTCALLs. 

@felipu is, I believe, supporting more than 64k in his project so he might have some acquired knowledge/thoughts

feilipu, 2019-01-07:
I've only implemented a scheme to manage banks of memory in 64kB chunks for a Z180, rather than a more general solution. The Z180 memory banking scheme is pretty well documented, so I'll only refer to [how I've set it up in my case](https://github.com/feilipu/yaz180#memory--io-address-layout) rather than the more general solution.

Supporting 64kB chunks means that there are substantial simplifications in calculating the MMU address within the Z180 world. Basically only the BBR Register needs to be touched in this case, so bank swapping can be quite a few cycles faster.

As part of the [logical solution for memory mapping](https://github.com/feilipu/yaz180/tree/master/yabios#memory-map), I've implemented [bios calls](https://github.com/feilipu/yaz180/tree/master/yabios#system-function-definitions) that help applications to call functions in arbitrary memory [`_call_far()`](https://github.com/feilipu/yaz180/tree/master/yabios#rst10defwdefb-_call_far), and to copy arbitrary memory in the 1MB address space [`_memcpy_far()`](https://github.com/feilipu/yaz180/tree/master/yabios#_memcpy_far). Setting banks of memory `_memset_far()` was written then bumped out a few commits ago, due to space constraints in the Common Area.

`RST` short calls are used to drive the `_call_far` functionality, as well as a few other necessary system calls. The code is all present in the [YAZ180 target in z88dk](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/target/yaz180). But to make it easier for me to maintain the sections and code size, it is [duplicated into a single file](https://github.com/feilipu/yaz180/blob/master/yabios/asm_common1_driver.asm) within the [YAZ180 repository](https://github.com/feilipu/yaz180/).

Where am I at with this far memory support now? Well I've got all the pieces I need in place. I can write applications that call arbitrary functions anywhere in memory using short calls, and I can safely copy memory (transparently l->r & r->l) from anywhere to anywhere.

I've used these functions to implement shell for loading applications into arbitrary banks, and a CP/M system that can be loaded from flash memory into any bank of RAM memory and uses the FATFS drives. Currently I'm working on a MP/M system that will use 7 banks of memory also using FATFS drives. Not much work has has happened on this in the past few months, but I hope to get back to MP/M and complete it shortly.

obiwanjacobi, 2019-01-08:
Not sure what SHORTCALLs are (I guess you mean RST calls, but is this part of z88dk?) [but I have got a beginning (untested yet)](https://github.com/obiwanjacobi/Zalt/blob/master/Source/z88dk_target/libsrc/_DEVELOPMENT/target/zalt/system/memory.far.asm). 
I am finding writing assembly a bit of a struggle. Seems like every time I re-read a piece of asm code I wrote I find something wrong with it.
I also deal with a hardware glitch in the MMU I need to fix first, that is why the software hasn't progressed much. Almost there though, I think.

Not sure about the use of `rst defb` short calls yet. Seems like the compactness at the caller site is does not weigh against unraveling the stack in the function itself. I am not shooting for most compact code, I am shooting for fasted code.

Thanks for your story, feilipu. I do not have bin or hex loaders yet and certainly no drives or file systems. I am still at the start of this journey and it is certainly good to know there is someone here with some actual experience.

Keep you posted.

obiwanjacobi, 2019-01-09:
[I think I have a first working test](https://github.com/obiwanjacobi/Zalt/blob/master/Source/z88dk_target/_tests/MemoryManager/FarMemory_test.c).

The test uses the MemoryManager API to manually switch (push) page 20 into memory (at 0x8000) and write a byte (42) into that page. Then the memory layout is restored (pop) and the farptr to that byte is returned from a function and later on dereferenced and 42 is printed.

Next is writing data - [I need another setup method for dealing with the exx regs etc...](https://github.com/obiwanjacobi/Zalt/blob/master/Source/z88dk_target/libsrc/_DEVELOPMENT/target/zalt/system/memory.far.asm)

feilipu, 2019-01-09:
BTW, I only mention it because you mention wanting the fastest code, so doing a [`call my_function`](https://github.com/obiwanjacobi/Zalt/blob/master/Source/z88dk_target/libsrc/_DEVELOPMENT/target/zalt/system/memorymanager.asm#L206) then `ret` in z80 assembly is unnecessary.

Just do `jp my_function` on the last function in the routine. You'll save 7 clocks using the `jp` (`jp` is 10, rather than 17 clocks), and then you'll save another 10 clocks on the redundant `ret`. That will soon add up.

:nerd_face:

obiwanjacobi, 2019-01-09:
Yes I knew that trick and use in [my C thunks](https://github.com/obiwanjacobi/Zalt/blob/master/Source/z88dk_target/libsrc/_DEVELOPMENT/target/zalt/system/memorymanager.thunk.Page_SetAt.asm) when there is no post processing.

There are so many things to think about when writing asm, that I tend to focus on functionality. Also I usually do see the best solution at first sitting. I regularly revisit asm code I've written to see if I think differently now ;-)

Thanks anyway, I am interested in every trick in the book.

feilipu, 2020-01-22:
Ok so far?

closed 2020-01-22 by feilipu