z88dk/z88dk #1062: Initialisation of 2d-array fails

boert, 2019-01-13:
On of the last updates broke my program.
Example code:
```c
#include <stdio.h>

unsigned char font[2][2] =
{
    { 0, 1},
    { 2, 3}
};

void main( void)
{
    printf( "%d", font[1][1]);
}
```

With git-bisect I found the last working commit:

git bisect good cc7f7cb27a3291fe7bdda9d096cf12420ce27087
good: [cc7f7cb27a3291fe7bdda9d096cf12420ce27087] +z88: Move scoping to the right files

git bisect bad e2238315f05c1bd8109777ed8707884f6a0b08a7
first bad commit: [e2238315f05c1bd8109777ed8707884f6a0b08a7] Issue #1054: Fix some initialisation issues

```
Error:
zcc +kc85 -create-app --list --c-code-in-asm  -o TEST test.c
sccz80:"test.c" L:7 Error:Missing token, expecting } got ,
sccz80:"test.c" L:9 Error:Missing token, expecting ; got {
make: *** [test] Error 1
```

closed 2019-01-13 by suborb

suborb, 2019-01-13:
Thank you for the report and apologies for missing that case.

boert, 2019-01-13:
No problem :)
Thank you for your very fast response!

commit aaae384b (2019-01-13): Fix #1062: Fix char array initialiation with constexpr, not strings