z88dk/z88dk #1063: patch: appmake/kc generate full 128 byte blocks

boert, 2019-01-13:
Some loading algorithms for KC85/2-5 rely on complete 128 byte blocks (tape, M052/USB).
This patch fill up a generated file to the correct size.

```
---
 src/appmake/kc.c | 7 +++++++
 1 file changed, 7 insertions(+)

diff --git a/src/appmake/kc.c b/src/appmake/kc.c
index e7eb7a7..46c802f 100644
--- a/src/appmake/kc.c
+++ b/src/appmake/kc.c
@@ -145,6 +145,13 @@ int kc_exec(char* target)
         writebyte(c, fpout);
     }
 
+    // fill the block to 128 bytes
+    while( (i % 128) > 0)
+    {
+        writebyte(0, fpout);
+        i++;
+    }
+
     fclose(fpin);
     fclose(fpout);
 
-- 
```

suborb, 2019-01-14:
Thanks - I've applied the diff.

If you have any more KC tweaks then they'll be gratefully received -  I'm sure there must be quicker ways to deal with the weird screen layouts!

closed 2019-01-14 by suborb

boert, 2019-01-17:
Hello suborb!

Thanks for applying my small patch!

Am 14.01.2019 um 22:44 schrieb suborb <notifications@github.com>:
> Thanks - I've applied the diff.
> 
> If you have any more KC tweaks then they'll be gratefully received - I'm sure there must be quicker ways to deal with the weird screen layouts
> 
Can you go give some more details for the problem?

I have some real hardware ready (KC85/5)
Some other computers are mothballed (Z9001, KC85/3, KC85/4).

The Z9001, KC85/1 (same like Z9001) and KC87 are produced by Robotron. The have only pseudo graphics.
A lot of them have even no color RAM, so the output is black&white. There exists some graphic extensions but I they are very rare.

The HC900 (same as KC85/2), KC85/3 and KC85/4 are produced by VEB Mikroelektronik „Wilhelm Pieck“ Mühlhausen.
VEB means volkseigener Betrieb (=nationally owned company). They have 320x200 pixel color graphics ex factory.
So they where much more used for games and where more common in private households.

The Robotron computers and the Mühlhausen computers have only the same BASIC ROM (BM600), if exists.

The KC85/5 is a KC85/4 with more RAM (+192k) and more ROM (+28k). The PCB was prepared for such enhancement, so I just had to pull out the old ROM and RAM, solder in IC sockets and put in bigger RAM chips and new EPROMs. 
The monitor program (called CAOS) got recently to version 4.7 and try to be compatible with the original version from KC85/4 (COAS 4.2).
But some programs (mostly games) try to manage hardware by there self, which can cause trouble. 

Maybe you can give write permission to my github account (boert) and I can do some small improvements by myself.

I wonder why you have written a new graphics driver for the system (generic_console.asm)?
And I wonder also why you bring a new font to the system?
The CAOS support user defined fonts via the CCTL0-3 addresses (0xB7A6-0xBA7C).

Best regards,
Bert