z88dk/z88dk #1069: (classic) TS2068 mode switching

suborb, 2019-01-20:
The TS2068 screen driver should support switching to the Timex hires and high colour screens.

ped7g, 2019-01-22:
JFYI:

On ZX-Next the Timex modes and port 0xFF are supported too (including the extra post-processing of colours by ULANext), but you have to set the bit 2 in Next Reg 0x08 "Peripheral 3 setting".

Also on Next the Timex mode is recommended way for double-buffering of classic mode, as using the ZX128k shadow bank will disable Layer-2 visibility due to memory chips timings (128k bank used for shadow VRAM is slightly slower than the ZX48/Timex bank used for 0x4000..0x7FFF).

aralbrec, 2019-01-25:
All ports are fully decoded in timex machines.

closed 2019-03-28 by suborb