z88dk/z88dk #1072: [classic] enhancing/fixing the Spectrum ROM subtype

zx70, 2019-01-22:
I'm extending the new function your text scrolling with a ROMless variant (useful to build a ROM).
the console drivers will then have an inclusion to this piece of code which will vary depending on the target (spectrum/ ts2068) and the ROM/RAM modes.

zx70, 2019-01-22:
hope I'm not colliding with #1069

zx70, 2019-02-06:
this is also related to #244, ROM calls are now mostly corrected for the ts2068

zx70, 2019-02-08:
The console was too crap, I'm reopening the issue and sorting out the things a little more.
The interrupt driven keyboard scanning is not any longer necessary because the single functions (e.g. getk(), fgetc_cons()) can be overriden where necessary, so I'm removing it.   The free bytes in ROM between location 8 and 56 can be used for the scroll routine.

zx70, 2019-02-11:
zcc +zx -lndos -O3 -subtype=rom -create-app -DMINIMALISTIC -DTINYMEM -DGRAPHICS -llib3d clisp.c
ROM version of the Campus Lisp able to run on a 16K Spectrum  :)

zx70, 2019-02-12:
I'm lost on a last detail missing, hope @suborb can help with it.
Few programs are not detecting the CR key correctly and generally get stuck (e.g. the othello program is not finalizing the player moves, the CR key is not translated to CR/LF, etc.), which makes me think at a wrong setting on console behavior.    Would you be able to spot the problem ?

suborb, 2019-03-28:
Sorry, just spotted this now - input/fgetc_cons_inkey is the routine to lose - I've redirected fgetc_cons_rom.asm to it.

zx70, 2019-10-09:
Closing this issue,  I can't remember whether there was still something missing but it looks reasonably complete.

closed 2019-10-09 by zx70