z88dk/z88dk #1075: (classic) Toshiba Pasopia 7 support

suborb, 2019-01-24:
Summary:

* z80 @ 4Mhz
* 64k RAM
* MC6845 
* 64k VRAM?? RG layers + G shared with text +  attribute?
* 2 x SN76489 + 1 bit sound
* Floppy drive
* 2 joysticks

_Gathering information stage_ 

Machine boots into BASIC, looks very similar to Multi8 (with mon command) so might be able to load files in same format.

suborb, 2019-01-24:
The machine can boot from a floppy disk to $d000. So a bootstrap can be located there.

Code in bios.rom at $40f9 is I think responsible for this. It cycles round the 4 drives looking for something bootable. First byte has to f3 (di)

So now there's a branch where we can load 512 byte programs, and the first, slightly wonky one is:

[image]

zx70, 2019-01-25:
In this case BASCK helps a lot on the disassembly phase, especially if you create a CTL file and use it with skoolkit.

zx70, 2019-01-25:
https://github.com/z88dk/techdocs/blob/master/targets/pasopia/tbasic.ctl

..hand edited, still weak but the resulting html output is already surprising

suborb, 2019-01-25:
I think I’m going to go for 64k RAM mode on this one. The VRAM is paged in at $8000 which is a little awkward since that’s the start of RAM with BASIC paged in.

zx70, 2019-01-25:
ouch

zx70, 2023-07-18:
I found a cp/m disk image found in archive.org, converted it to IMD with the hxc tools and run IMDU.COM on it:
There must have been some error in the conversion tool, but apparently the first track is zeroed ??

IMageDisk Utility 1.18 / Mar 07 2012
IMD 1.17: 18/06/2023 11:14:46
File generated by the HxC Floppy Emulator software v2.14.4.1
Assuming 1:1 for Binary output
 0/0 250 kbps DD  8x512
      1   2   3   4   5   6   7   8
      D00 D00 D00 D00 D00 D00 D00 D00
 0/1  D   D   D   D   D   D   D   D
 1/0  D   D   D   D   D   D   D   D
 1/1  D   D   D   D   D   D   D   D
 2/0  D   D   D   D   D   D   D   D
 2/1  D   D   D   D   DE5 DE5 DE5 DE5
 3/0  D   D   D   D   D   D   D   D
 3/1  D   D   D   D   D   D   D   D
 4/0  D   D   D   D   D   D   D   D
 4/1  D   D   D   D   D   D   D   D
 5/0  D   D   D   D   D   D   D   D
 5/1  D   D   D   D   D   D   D   D
 6/0  D   D   D   D   D   D   D   D
 6/1  D   D   D   D   D   D   D   D
 7/0  D   D   D   D   D   D   D   D
 7/1  D   D   D   D   D   D   D   D
 8/0  D   D   D   D   D   D   D   D
 8/1  D   D   D   D   D   D   D   D
 9/0  D   D   D   D   D   D   D   D
 9/1  D   D   D   D   D   D   D   D
10/0  D   D   D   D   D   D   D   D
10/1  D   D   D   D   D   D   D   D
11/0  D   D   D   D   D   D   D   D
11/1  D   D   D   D   D   D   D   D
12/0  D   D   D   D   D   D   D   D
12/1  D   D   D   D   D   D   D   D
13/0  D   D   D   D   D   D   D   D
13/1  D   D   D   D   D   D   D   D
14/0  D   D   D   D   D   D   D   D
14/1  D   D00 D   D00 D   D00 D   D
15/0  D   D   D   D   D   D   D   D
15/1  D   D   D   D   D   D   D   D
16/0  D   D   D   D   D   D   D   D
16/1  D   D   D   D   D   D   D   D
17/0  D   D   D   D   D   D   D   D
17/1  D   D   D   D   D   D   D   D
18/0  D   D   D   D   D   D   D   D
18/1  D   D   D   D00 D   D   D   D
19/0  D   D   D   D   D   D   D   D
19/1  D   D   D   D   D   D   D   D
20/0  D   D   D   D   D   D   D   D
20/1  D   D   D   D   D   D   D   D
21/0  D   D   D   D   D   D   D   D
21/1  D   D   D   D   D   D   D   D
22/0  D   D   D   D   D   D   D   D
22/1  D   D   D   D   D   D   D   D
23/0  D   D   D   D   D   D   D   D
23/1  D   D   D   D   D   D   D   D
24/0  D   D   D   D   D   D   D   D00
24/1  D   D   D   D   D   D   D   D
25/0  D   DE5 D   DE5 D   DE5 D   DE5
25/1  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
26/0  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
26/1  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
27/0  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
27/1  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
28/0  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
28/1  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
29/0  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
29/1  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
30/0  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
30/1  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
31/0  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
31/1  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
32/0  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
32/1  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
33/0  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
33/1  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
34/0  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
34/1  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
35/0  D00 D00 D00 D00 D00 D00 D00 D00
35/1  D00 D00 D00 D00 D00 D00 D00 D00
36/0  D00 D00 D00 D00 D00 D00 D00 D00
36/1  D00 D00 D00 D00 D00 D00 D00 D00
37/0  D00 D00 D00 D00 D00 D00 D00 D00
37/1  D00 D00 D00 D00 D00 D00 D00 D00
38/0  D00 D00 D00 D00 D00 D00 D00 D00
38/1  D00 D00 D00 D00 D00 D00 D00 D00
39/0  D00 D00 D00 D00 D00 D00 D00 D00
39/1  D00 D00 D00 D00 D00 D00 D00 D00
40/0 No data
40/1 No data
41/0 No data
41/1 No data
84 tracks(42/42), 640 sectors (253 Compressed)

zx70, 2023-07-18:
I now understand that the resulting image is not good enough to boot on an emulator.
The pasopia7 one in the takeda's collection reacts differently when a valid blank disk image is loaded.

zx70, 2023-07-18:
To get a valid boot image:
imdu /D /B IL= X=40,41 ToshibaT100PersonalComputerCPM22CBASIC2Version171982_disk1of1_img.imd cpm.raw
bin2imd cpm.raw cpm17.imd DM=5 N=40 SS=256 SM=1-16 /2

zx70, 2023-07-18:
I'm not sure on how track0 is supposed to be, the blank image I used as a reference had  smaller sectors in track0, side0 and had 35 tracks.
This image is bigger, but  I'm sure it wasn't 40 tracks originally:

IMageDisk Utility 1.18 / Mar 07 2012
IMD 1.18: 18/07/2023 19:20:12
Assuming 1:1 for Binary output
cpm17.raw already exists - proceed(Y/N)?Y
 0/0 250 kbps DD  16x256
      1   2   3   4   5   6   7   8   9   10  11  12  13  14  15  16
      D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00
 0/1  D   D   D   D   D   D   D   D   D   D   D   D   D   D   D   D
 1/0  D   D   D   D   D   D   D   D   D   D   D   D   D   D   D   D
 1/1  D   D   D   D   D   D   D   D   D   D   D   D   D   D   D   D
 2/0  D   D   D   D   D   D   D   D   D   D   D   D   D   D   D   D
 2/1  D   D   D   D   D   D   D   DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
 3/0  D   DE5 D   D   D   DE5 D   D   D   DE5 D   D   D   DE5 D   D
 3/1  D   D   D   D   D   D   D   D   D   D   D   D   D   D   D   D
 4/0  D   D   D   D   D   D   D00 D   DE5 D   D00 D   DE5 D   D00 D
 4/1  D   D   D   D   D   D   D   D   D   D   D   D   D   D   D   D
 5/0  D   D   D   D   D   D   D   D   D   D   D   D   D   D   DE5 D
 5/1  D   D   D   D   D   D   D   D   D   D   D   D   D   D   D   DE5
 ...)
33/0  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
33/1  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
34/0  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
34/1  DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5 DE5
35/0  D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00
35/1  D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00
36/0  D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00
36/1  D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00
37/0  D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00
37/1  D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00
38/0  D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00
38/1  D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00
39/0  D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00
39/1  D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00 D00
80 tracks(40/40), 1280 sectors (549 Compressed)

zx70, 2023-07-19:
So, a better tuning must be:
\z80\teledisk\imdu /D /B IL= X=35,41 ToshibaT100PersonalComputerCPM22CBASIC2Version171982_disk1of1_img.imd cpm.raw
bin2imd cpm.raw cpm17.imd DM=5 N=35 SS=256 SM=1-16 /2

zx70, 2023-07-19:
Yet another interesting project with interesting information here:
https://github.com/beijingduckx/pasopia7_cpm22/tree/main

Note that the image I recovered boots on the takeda's emulators collection on both pasopia.exe and pasopia7.exe but in the latter case the 80 columns text is not visible.   The cursor helps understanding that the code is running properly, though.

zx70, 2023-07-20:
To create a disk image able to run config.exe:
- Get "pasopia.exe" from the Takeda's repository
- Mount cpm17.imd on drive A: and boot the system
- Use the emulator's menu to create and mount a "blank 2D disk", in D88 format in drive B:
- run FORMAT.COM
- Run the emulator's debugger (control/main CPU)
- In the debugger window, type "G 115D"
- run CONFIG.COM,  the debugger will stop just before an error condition

- To skip the error,  type "R A 0"
- type "G 115D" to repeat the loop

A must be set to zero, as in this abstract
<code>
done    0000115C  EX DE, HL
F = [------Z-]  A = 00  BC = 00
F'= [CN--H--S]  A'= 03  BC'= 00
</code>

- If it is zero, you finished skipping the bad sectors, type "G".   Otherwise type "G 115D" and repeat the 2 previous steps

- Once the program is ready. use option 6 to put the system in drive B
- run PIP b:=a:*.*  and copy all the files
- unmount the disk in drive b:

The emulator always loads the disks in IMD format as read-only, the resulting D88 disk will be R/W.