z88dk/z88dk #1076: z80asm: windows 10 install cannot generate binaries in linking step

aralbrec, 2019-01-25:
A windows install from nightly on a win10 machine is failing in the final linking step.  I don't have a win10 machine handy to test with so I'm hoping @pauloscustodio or someone with a win10 machine can check into this.

The failure occurs with the nightly binaries as well as my locally compiled vs2015 binaries.

Simple test.c:

```
#include <stdio.h>
 
main()
{
   printf("Hello World !\n");
}
```

Verbose output:

```
C:\Dev\z88dk>zcc +zx -v test.c -o test -lndos

PROCESSING test.c
ucpp -iquote"." -D__Z88DK -D__SPECTRUM -DSPECTRUM -D__SPECTRUM__ -D__Z80 -DZ80  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"C:\Dev\z88dk\lib\config\..\..\/include"    "test.c" "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.i2"
zpragma -sccz80 < "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.i2" > "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.i"
sccz80  -ext=opt -mz80 -standard-escape-chars   "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.i" -o "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.opt"
copt "C:\Users\Mcgibbitts\Dev\z88dk\lib\config\..\..\/lib/z80rules.9" < "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.opt" > "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.op1"
copt "C:\Users\Mcgibbitts\Dev\z88dk\lib\config\..\..\/lib/z80rules.2" < "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.op1" > "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.opt"
copt "C:\Users\Mcgibbitts\Dev\z88dk\lib\config\..\..\/lib/z80rules.1" < "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.opt" > "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.asm"
type "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.asm" >> "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.tmp"
type "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.tmp" >> "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.asm"
z80asm -I"C:\Dev\z88dk\lib\config\..\..\\lib\target\zx\def" -D__SCCZ80  -s  -mz80 -I"C:\Dev\z88dk\lib\config\..\..\/lib" "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.asm"

PROCESSING C:\Dev\z88dk\lib\config\..\..\\lib\target\zx\classic\spec_crt0.asm
type "C:\Dev\z88dk\lib\config\..\..\\lib\target\zx\classic\spec_crt0.asm" >> "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F3.asm"
type "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F3.asm" >> "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F3.tmp"
type "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F3.tmp" >> "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F3.asm"
WILL ACT AS CRT

GENERATING OUTPUT

PROCESSING CRT
z80asm -I"C:\Dev\z88dk\lib\config\..\..\\lib\target\zx\def" -D__SCCZ80  -s  -mz80 -I"C:\Dev\z88dk\lib\config\..\..\/lib" -I"C:\Dev\z88dk\lib\config\..\..\\lib\target\zx\classic"  "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F3.asm"

z80asm -mz80 -b -d  -o"test" -L.   -D__SCCZ80 -L"C:\Dev\z88dk\lib\config\..\..\/lib/clibs" -I"C:\Dev\z88dk\lib\config\..\..\/lib" -indos -izx_clib -i"z80_crt0"     "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F3.o" "C:\Users\Mcgibbitts\AppData\Local\Temp\zccA26F2.o"
test: Permission denied
[WARN ] (..\..\src\z80asm\lib\alloc.c:122 errno: Permission denied) memory leak (48 bytes) allocated at ..\..\src\z80asm\z80asm.c:254
[WARN ] (..\..\src\z80asm\lib\alloc.c:122) memory leak (12 bytes) allocated at ..\..\src\z80asm\z80asm.c:281
[WARN ] (..\..\src\z80asm\lib\alloc.c:122) memory leak (58 bytes) allocated at ..\..\src\z80asm\z80asm.c:254
[WARN ] (..\..\src\z80asm\lib\alloc.c:122) memory leak (12 bytes) allocated at ..\..\src\z80asm\z80asm.c:281
[WARN ] (..\..\src\z80asm\lib\alloc.c:122) memory leak (57 bytes) allocated at ..\..\src\z80asm\z80asm.c:254
[WARN ] (..\..\src\z80asm\lib\alloc.c:122) memory leak (12 bytes) allocated at ..\..\src\z80asm\z80asm.c:281
[WARN ] (..\..\src\z80asm\lib\alloc.c:122) memory leak (54 bytes) allocated at ..\..\src\z80asm\z80asm.c:254
[WARN ] (..\..\src\z80asm\lib\alloc.c:122) memory leak (12 bytes) allocated at ..\..\src\z80asm\z80asm.c:281
[WARN ] (..\..\src\z80asm\lib\alloc.c:122) memory leak (8 bytes) allocated at ..\..\src\z80asm\z80asm.c:276

C:\Dev\z88dk>
```

As you can see, z80asm only fails in the final linking step.  zcc is able to create the "zcc_opt.def" file in the output directory so I don't think it's because there is no write permission in the output dir.  Varying the compile location to different directories did not help.

pauloscustodio, 2019-01-25:
There is a "test" directory in z88dk root, therefore z80asm cannot create the output file named "test".

Please try to change the output file name:
```
zcc +zx -v test.c -o test.bin -lndos
```

aralbrec, 2019-01-26:
That was it, silly error

closed 2019-01-26 by aralbrec