z88dk/z88dk #1077: Crazy idea for speed optimization

zx70, 2019-01-25:


zx70, 2019-01-26:
..thinking back to the 'jr<>jp' substitution, it could stay even in the assembler as an option,, the IXIY trick is insane but very handy...

pauloscustodio, 2019-01-27:
While easy to do, does it make sense?
- JR = 12 cycles, JP = 10 cycles - so 2 cycles won at the expense of 1 byte
- JR flag = 7 cycles on no jump, 12 on jump; JP flag = 10 cycles - so 3 cycles lost or 2 won, at the expense of 1 byte
- DJNZ = 8 cycles on no jump, 13 on jump; DEC B/JP NZ = 14 cycles, always worse, at the expense of 2 bytes

pauloscustodio, 2019-01-27:
JR-JP done, Please test and give some feedback.

obiwanjacobi, 2019-01-27:
And if the code is to be relocatable, can you replace JP's with (multiple) JR's?

zx70, 2019-01-27:
please don't ask it  :)   a range checking would be needed. .

zx70, 2019-01-27:
@obiwanjacobi ,  issue #13 discussed how to evolve the self reloc featurea of z80asm

zx70, 2019-01-27:
..hmm  I'm quickly changing my mind:
https://www.cpmspectrepi.uk/z80/z80.htm

zx70, 2019-01-28:
I'm seriously thinking at preparing an opt ruleset to convert the z80 specific instructions to calls to equivalent 8080 compatible subroutines.  It is prone to any time of criticism (which we are already used to hear)  but, if working,  it will be a simple bridge to a new cpu architecture (8080, 8085).   Sounds crazy but it is probably less painful than a 486 port  :)

zx70, 2019-01-29:
I tested the new option on both the libraries  (by changing Make.config) and on the zcc command line using the zx81 target to perceive a speed difference due to its slowness.
In the latter case something often breaks (e.g. on dstar.c) and I still haven't understood what happens exactly, but the rebuilt library works well and seems to me slightly faster.  The program sizes are very close, though (only 1 byte bigger !), in most of the comparisons I've done the size varies much more.

zx70, 2019-01-29:
Two extra opt rules found, I'm adding them considering them not dangerous because similar ones were applied already.

	scf
	jr	z,ASMPC+3
	ccf
	jp	c,%1
=
	jp	z,%1

	scf
	jr	z,ASMPC+3
	ccf
	jp	nc,%1
=
	jp	nz,%1

suborb, 2019-01-29:
Be careful! The carry may be converted into an int - eg when you assign to a variable the result of a comparison.

zx70, 2019-01-29:
Also:
	jp	z,%1
	ld	hl,%3	;const
	jr	%2
.%1
	ld	hl,%4	;const
.%2
	ret
=
	ld	hl,%3	;const
	ret	nz
.%1
	ld	hl,%4	;const
.%2
	ret

zx70, 2019-01-29:
Thankyou, this is the kind of feedbak I hoped to get.

zx70, 2019-01-29:
Comments on this one ?
	jp	z,%1
	ld	hl,%3	;const
	jr	%2
.%1
	ld	hl,%4	;const
.%2
	ret
=
	ld	hl,%3	;const
	ret	nz
.%1
	ld	hl,%4	;const
.%2
	ret

ped7g, 2019-01-29:
should be correct... although makes me wonder why even the first `jr %2` is not just `ret`, i.e. where the original unoptimized code comes from... probably some building block not aware that beyond `%2` comes `ret` instantly. But your optimization is correct only when that is true.

zx70, 2019-01-29:
I'm keeping the previous one and:

        jp    z,%1
    .%1
        ret
    =
    .%1
        ret

I'm now too scared to insist.

suborb, 2019-01-29:
> Comments on this one (edited)?
> 
> ```
>     jp    c,%1
>     ld    hl,%3    ;const
>     jr    %2
> .%1
>     ld    hl,%4    ;const
> .%2
>     ret
> =
>     ld    hl,%3    ;const
>     ret    nc
> .%1
>     ld    hl,%4    ;const
> .%2
>     ret
> ```

You can remove the ;const markers, it's valid without them (eg when picking up from a constant variable)

suborb, 2019-01-29:
> should be correct... although makes me wonder why even the first `jr %2` is not just `ret`, i.e. where the original unoptimized code comes from... probably some building block not aware that beyond `%2` comes `ret` instantly. But your optimization is correct only when that is true.

sccz80 mostly works a statement at a time so it's unaware a return is following. It's on my list of things to look at - building a syntax tree and then optimising on that. 

These cases are fairly simple because there's no local variables. But generally %2 will contain stack cleanup as well, so many times the jr/jp makes sense to cut down program size.

zx70, 2019-01-29:
Thankyou Dom, I'm doing it immediately.   I'm not adding further rules, but I'm now thinking back at the @obiwanjacobi request, possible optimizations on the following code pieces like :

    	jp	nc,i_117
    	ld	hl,1	;const
    	jr	i_118
    .i_117

..or similar (e.g. with "jp nz.."), suggest a possible JP<>JR replacement, because proven to be in a valid range.

I wonder if a JP<>JR substitution conditional to the actual jump distance is possible on Z80ASM ?

zx70, 2019-01-29:
Computing the benefit of substituting JP with JR.     I tried adding the following rules to 'z80rules.0' and compared the results (using the -O3 option, for max opt.):

		jp	%1
	%2
	%3
	%4
	%5
	%6
	%7
	%8
	.%1
	=
		jr	%1
	%2
	%3
	%4
	%5
	%6
	%7
	%8
	.%1

		jp	%1
	%2
	%3
	%4
	%5
	%6
	%7
	.%1
	=
		jr	%1
	%2
	%3
	%4
	%5
	%6
	%7
	.%1

		jp	%1
	%2
	%3
	%4
	%5
	%6
	.%1
	=
		jr	%1
	%2
	%3
	%4
	%5
	%6
	.%1

		jp	%1
	%2
	%3
	%4
	%5
	.%1
	=
		jr	%1
	%2
	%3
	%4
	%5
	.%1

		jp	%1
	%2
	%3
	%4
	.%1
	=
		jr	%1
	%2
	%3
	%4
	.%1

		jp	%1
	%2
	%3
	.%1
	=
		jr	%1
	%2
	%3
	.%1

..othello.c gets 27 bytes smaller, adv_a 16 bytes smaller.   I expect that by extending the range checking a reasonable memory space can be recovered.

Obviously not all the conditional JP calls have an equivalent JP instruction and guessing the exact lenght of any jump before a first substitution pass would be rather difficult

zx70, 2019-01-29:
By insisting and substituting also conditional jumps up to 8 rows forward and backwards already reduces othello.c by 361 bytes.

pauloscustodio, 2019-01-29:
> Paulo, could it be simpler to add a foo instruction doing the relative jump and switching to absolute in case of overflow ?

There is no need for a special instruction; any JR/DJNZ to a label too far should be automatically converted to JP. This is item 18 of the wish-list (https://github.com/z88dk/z80asm/wiki).

Although easy to describe, it cannot be implemented in the current z80asm without a major change, as it now builds the binary code one instrution at a time, and it would be hard to insert one byte at address-resolution time to replace the JR by a JP.

It is planned for z80asm2.

zx70, 2019-01-29:
I understand, thanks for all the support  :)
I'm keeping this topic open because I'm not understanding the random instability I'm getting with your new option.

obiwanjacobi, 2019-01-30:
> ```
>     jp    z,%1
> .%1
>     ret
> =
> .%1
>     ret
> ```

More general:
Any JP, JR with or without condition, can be removed if the jump target (label) is at the next instruction address.

pauloscustodio, 2019-01-30:
> Any JP, JR with or without condition, can be removed if the jump target (label) is at the next instruction address.

Any CALL follwed by RET can be replaced by JP?

zx70, 2019-01-30:
What I discovered is that almost all this kind of optimizations are already present (I should have checked everything with the -O3 flag enabled).
The only relevant set of rules I've found is the JP->JR one, but I'm afraid that on the current optimizers it will be too messy.

obiwanjacobi, 2019-01-30:
> > Any JP, JR with or without condition, can be removed if the jump target (label) is at the next instruction address.
> 
> Any CALL follwed by RET can be replaced by JP?

Only if you control the stack completely and KNOW it is balanced at all times.

I have an example that will fail this here: 
https://github.com/obiwanjacobi/Zalt/blob/master/Source/z88dk_target/libsrc/_DEVELOPMENT/target/zalt/system/memory.far.asm

The setup calls leave extra data on the stack that is used and cleaned up by the teardown calls. If you would replace the call - ret sequences in the lp_ implementations it goes of into hyperspace.

zx70, 2019-01-30:
I get the point.   Luckily the compiler output is a bit more predictable ( well, it usually is.. see the Dom's warning on top of this thread  :S ).
Nevertheless all the simple gaps were filled long ago squeezing out all the smartness we had for them..  then, again, putting SDCC in the loop was an opportunity for a revision.   We still find something here and there but the recent changes were intended to remove nasty optimizations due to arising bugs rather than inserting new ones.

zx70, 2019-02-06:
the 8080 topic was already discussed on #339.   my opinion non is that rather than native 8080 the z80asm assembler could try to emulate the missing operations

suborb, 2019-02-07:
> the 8080 topic was already discussed on #339. my opinion non is that rather than native 8080 the z80asm assembler could try to emulate the missing operations

I think we'd need some sort of compiler support. sccz80 uses z80 opcodes (eg shifts) quite extensively. Might be worth looking to see what sort of code ACK produces.

suborb, 2019-08-09:
The 8080 thread has been tackled and the jr/jp replacement is in the backlog for z80asm2 so I'm going to close this issue now.

closed 2019-08-09 by suborb