z88dk/z88dk #1080: [pasopia7] No pasopia7.cfg

Fabrizio-Caruso, 2019-01-27:
I am aware that the Pasopia 7 is a work in progress but I was expecting to find the +pasopia7 target option mentioned in the manual.

suborb, 2019-01-27:
I'm not sure where you're looking, but the Pasopia7 page is here:

https://github.com/z88dk/z88dk/wiki/Platform---Toshiba-Pasopia7

and it's on the platform list here:

https://github.com/z88dk/z88dk/wiki/Platform

suborb, 2019-01-27:
And the config file is here: https://github.com/z88dk/z88dk/blob/master/lib/config/pasopia7.cfg

Whilst being brought up, development was in a branch which got merged in this PR: https://github.com/z88dk/z88dk/pull/1079/files

Fabrizio-Caruso, 2019-01-27:
@suborb  I did not find it in the latest snapshot. I need to double check...

suborb, 2019-01-27:
Ah, I only merged it this evening so it won't have made the build last night!

closed 2019-01-27 by suborb