z88dk/z88dk #1084: (appmake) Add FAT image writer support

suborb, 2019-01-30:
This will allow us to create discs for MSX or other platforms.

I can't face writing yet another FAT driver, so tempted to import fatfs and point it to the disc image container that we use for CP/M discs.

Should probably separate out the CP/M filesystem from the disc image container code as well.

feilipu, 2019-01-30:
This version of [ChaN FATfs](https://github.com/feilipu/z88dk-libraries/tree/master/ff) has been slightly customised for z88dk, with some little ended optimisation by @aralbrec.

One time I tried to detangle it into separate files, but it quickly became unmaintainable.
ChaN does patch and update it often, and it would be a shame to make patching more complex than it is currently.

It takes an age to compile though...

suborb, 2019-01-30:
This will be for appmake so we can take the vanilla version without any modifications since it won't be compiled for the z80.

Having support for FAT16 + 32 may well be useful for creating memory cards for some targets.

suborb, 2019-02-02:
Implemented, so far only for MSXDOS disc format (FAT12) and only minor modifications needed to fatfs

closed 2019-02-02 by suborb