z88dk/z88dk #1089: -subtybe=dsk zcc option failing

FlorentFlament, 2019-02-03:
Hello,

A couple of weeks ago, I was able to generate a `.dsk` CPC disk image using the following `zcc` command. Recently it has been failing with the following error message:

```
$ zcc +cpc -lndos -subtype=dsk -create-app -o helloc hello_c.c
cpc: Can't write disc imageBuilding application code failed
$
```

Is this a regresion ?

Here's the `hello_c.c` that has been used:
```
$ cat hello_c.c 
#include <stdio.h>

int main() {
    printf("Hello World !\n");
    return 0;
}
```

Regards.
Florent

closed 2019-02-03 by suborb

suborb, 2019-02-03:
Sorry, I  must have broken it when I added +3 disc support.

Hopefully repaired now, thanks for reporting.

FlorentFlament, 2019-02-03:
Yep, this is fixed. That was fast ; )
Thanks !