z88dk/z88dk #1096: (classic) Improve Kaypro ii graphics

suborb, 2019-02-06:
[From RobertK issues email]

The gfxkp library cannot pointxy() and gfxkp83 restricts program size (since it has an ORG at 32k)

However, using the same trick as on the tiki100 (copying code to high memory) we can access the graphics page without any limitation on program size.

Since we can do that, we can easily get gencon working as an option.

We can then get the graphics working - I'm wonder if it's better just to print chr$(31) and give a simplified resolution of 80x25 rather than the interesting looking 80x50 that's currently available.

As reference the x50 vertical resolution is achieved by printing:

```
`
,
|
```

for the vertical pixels, though on the Kaypro the pipe is split in two and backtick is stronger.

zx70, 2019-02-07:
Be aware of a bug in the MAME emulator, since the graphics symbols couldn't cover all the pixel combinations the graphics drivers have to deal with text attributes and invert the characters.  
This is not correctly implemented in MAME:
[image]

suborb, 2019-02-07:
Thanks Stefano, I'm looking at the KayproII at the moment which just has ascii, greeks and a blinking characters > 128 at the moment. So that's a new replacement for gfxkp83.

I've not looked at gfxkp yet  - I don't think I've got the appropriate ROMs and boot discs. When I do I'll create a new subtype for it to avoid the confusion that I went through last night.

zx70, 2019-02-07:
I think that the Kaypro10 (84) is capable of its own graphics, I don't remember whether it is compatible to the Kaypro4, though.
[image]
Comment taken from the BIOS source (29-Mar-83);
SPECIFICATIONS:
	1) Using the screen as a 100 by 160 pixel matrix, this program
is to turn individual pixels in the matrix on and off.
	2) Normal ASCII characters (ASCII codes 0 to 127H) are not to
be written over by graphics characters, with the exception of SPACE,
or BLANK, characters. (ASCII code 20h)
	3) Pixels are numbered from 0 to 99 vertically and from
0 to 159 hrizontally, with 0,0 being the upper left corner of the
screen and 99,159 being the lower right corner of the screen.
	4) An offset is added to each pixel coordinate, said offset
typically the same as that used to send cursor addressing coordinates,
the points for the line routines, etc.
	5) Pixels with Illegal point coordinates (off screen) are ignored.
	6) The pixels are arranged in 4 by 2 cells, and the cells arranged
in a 25 by 80 grid to make up the screen.
 	_____
	|1|0|
	|3|2|
	|5|4|
	|7|6|
	-----

zx70, 2019-02-07:
[kaypro-graphics.pdf](https://github.com/z88dk/z88dk/files/2840226/kaypro-graphics.pdf)

suborb, 2019-02-07:
I think you're right, looking again at Mame source code, the KayPro 10 is the Kaypro 4/84 with a floppy drive removed. Table updated.

suborb, 2019-02-07:
I've renamed -subtype=kayproii to -subtype=kaypro83 and added a -subtype=kaypro84.

I've left vpeek unimplemented since the emulators aren't working properly

closed 2019-02-07 by suborb