z88dk/z88dk #1097: (classic) Attache: plotted points not visible

suborb, 2019-02-07:
[From RobertK email]

"Everything works, but plotted points not actually seen".

I can't remember the code, but suggests that the paging (if there is any) is wrong and we're hitting main memory not VRAM.

zx70, 2019-02-07:
Odd,  Fabrizio confirmed that the new drivers were working,  I'll check it.

zx70, 2019-02-08:
[image]
I tried with globe.c and coswave.c.. everything works as expected: pixel drawing, line drawing and "classic" sprites,

suborb, 2019-02-08:
That looks good, thanks for checking it out.

closed 2019-02-08 by suborb

zx70, 2019-02-09:
I think I understood what happened:  a missing clg(); at the beginning to enter ingraphics mode?