z88dk/z88dk #1098: (classic) Enterprise graphics not compiling

suborb, 2019-02-07:
[From RobertK email]

Enterprise 64/128: I assume I need to specifiy either the -gfxep or -lgfxephr graphics library.
When compiling it like this

    zcc +enterprise -gfxep -create-app -o pte plottest.c 

I get compiler errors "symbol 'plot' not defined" (and it also complains about unplot, c_plot, getmaxx, etc.)
And like this

    zcc +enterprise -lgfxephr -create-app -o pte plottest.c 
f
there are fewer error messages, but they still say that point() and c_plot() were not defined.

zx70, 2019-02-28:
The problems with point() and c_plot() are expected, the current driver does not interact at low level with the video memory, it just sends the drawing commands to the console handler.
I'm troubleshooting the first part.

zx70, 2019-02-28:
Is the test program available?

suborb, 2019-02-28:
It looks like it's just w_point() that's missing with -lgfxep

[plottest.txt](https://github.com/z88dk/z88dk/files/2916747/plottest.txt)