z88dk/z88dk #1099: (classic) Pacman Tile Maker

suborb, 2019-02-07:
[From RobertK email]

The Pacman tileset is not complete and not all graphic points are available.

It would be nice if we could convert a standard 8x8 font into a ROM suitable for the Pacman hardware.

zx70, 2019-02-15:
I thought a little at it,  we have few interesting configurations  possible: mixed text and graphics symbols,  all monochrome graphics symbols to push the resolution at the maximum possible (not much to gain, though), and low rez graphics symbols chosen in a way permitting a bit of colour shuffling.  I think a simple static font including symbols on a 2x3 matrix will suffice, easy variants could provide coloured text.

zx70, 2019-02-21:
[image]

This was obtained by converting FONT8.BIN (2048 bytes) still present in the 'ansi' folder in libsrc

suborb, 2019-02-21:
@zx70 thanks for doing that, now we've got the basics we can do a bit more!

zx70, 2019-02-21:
yup, I'm extending the tool to stick the 3x2 graphics symbols in the converted font, it takes 64 characters, there's space for 2 copies encoded in different colours but I'm not sure.   I wonder if it is worth to add an option to merge two fonts in a coloured one

suborb, 2019-02-21:
Nice. I was mulling over shifting to a standard ascii range when running with a custom font. 

I think we should be able to merge fonts - eg combine a udg style graphics with a text font and blocks as needed.

zx70, 2019-02-25:
[image]

zx70, 2019-02-25:
This is only a quick and dirty test of the font, I can leave Dominic the lib part  :P    This new rez (with a better driver than this mockup) sould be 84x72.
The 3x2 blocky graphics characters are between 192 and 255,  the conversion tool at the moment deals with fonts encoded from chr(0) to chr(127), eventual extra data is overwritten with the graphics data.

zx70, 2019-02-25:
I put a prebuilt font in the support/pacman folder

zx70, 2019-02-26:
I just discovered one of the limits of the driver I used in the tests:   the div3 table was too short to work on 84 values..

zx70, 2019-02-26:
Graphics driver updated, it now works at a slightly higher resolution (84x72) but the ROM font currently present in support/pacman must be provided.

zx70, 2019-02-26:
New graphics driver is now ready, I wouldn't waste time and energy to provide a 4x4 font.   A graphics only font covering the whole 256 characters range could push the resolution to 112*72 (4x2 pseudo-graphics character blocks), but there wouldn't be space left for text characters.    I'm not touching the console driver, probably it is easier for Dom to do the magic with the new console skeleton  :)

closed 2019-02-26 by zx70