z88dk/z88dk #1102: (sccz80) Initialising an array of structures fails.

thefossilrecord, 2019-02-09:
Updating to the latest version of sccz80 on Linux (14110-9df52a2-20190208) breaks existing code.

```
// example.c
#define EXAMPLE_SIZE	8

struct level
	{
	unsigned char tiles[EXAMPLE_SIZE];
	unsigned char data1[EXAMPLE_SIZE];
	unsigned char data2[EXAMPLE_SIZE];
	};

#define MAX_LEVELS	3
	
struct level level_data[MAX_LEVELS] =
	{	
		// Level 1
		{
		"\x01\x01\x01\x01\x01\x01\x01",
		"foo",
		"bar!",
		},
		// Level 2
		{
		"\x01\x01\x01\x01\x01\x01\x01",
		"foo",
		"bar!",
		},
		// Level 3
		{
		"\x01\x01\x01\x01\x01\x01\x01",
		"foo",
		"bar!",
		},		
	};

int main()
	{
	return 0;
	}
```

Compiling generates the following:

```
zcc +zx -vn -startup=31 -clib=new example.c

sccz80:"example.c" L:19 Error:Missing token, expecting } got ,
sccz80:"example.c" L:22 Error:Missing token, expecting ; got {
sccz80:"example.c" L:23 Error:Missing token, expecting ( got \
sccz80:"example.c" L:23 Error:Illegal Argument Name: 
sccz80:"example.c" L:23 Error:Illegal Argument Name: 
sccz80:"example.c" L:23 Error:A parameter named x01 has already been defined for function
sccz80:"example.c" L:23 Error:A parameter named x01 has already been defined for function
sccz80:"example.c" L:23 Error:A parameter named x01 has already been defined for function
sccz80:"example.c" L:23 Error:A parameter named x01 has already been defined for function
sccz80:"example.c" L:23 Error:A parameter named x01 has already been defined for function
```

This compiles without issue with an older version of sccz80 (9376-db832f0-20170827) and zsdcc ( 3.8.5 #10892).

suborb, 2019-02-09:
Ah, it's not coping with the trailing comma.

suborb, 2019-02-09:
I've just added a fix to allow for trailing commas in struct initialisation. Thanks for the report.

closed 2019-02-09 by suborb

pauloscustodio, 2019-02-14:
The test Issue_1102_struct_trailing_comma.c is still failing in my setup (MINGW64_NT-10.0)

Is this my setup's fault?

suborb, 2019-02-15:
Does the nightly build work?

pauloscustodio, 2019-02-19:
Solved with clean install of mingw

commit 9a12e685 (2019-02-09): Fix #1102: Handle a trailing comma in struct initialisation