z88dk/z88dk #1104: (kit) Rename binaries to start with z88dk-

suborb, 2019-02-10:
It looks like we're now experiencing clashes with ucpp as well as previously experienced ones with z80asm.

So lets prefix all binaries (apart from zcc) with z88dk- 

At present this is an install option $(EXEC_PREFIX) which won't work too well for in-tree development since it's passed down from the top level makefile.

suborb, 2019-02-10:
The following now have a z88dk- prefix:

* copt
* ucpp
* zcpp
* zpragma
* basck
* z80svg
* font2pv1000
* zx7/dzx7
* ticks

Currently unchanged:

* sccz80
* z80asm
* z80nm
* zobjcopy
* appmake

Ones that won't change:

* zcc
* zsdcc
* zsdcpp

pauloscustodio, 2019-02-13:
How about z88dk-asm instead of z88dk-z80asm?

suborb, 2019-02-13:
I'd be happy with z88dk-z80asm - that's what it's known as for some .deb packages I've seen.

Unfortunately it's a tool that people call directly and changing the name will break everyone who does that. The ones I've changed so far are either just internal tools, or "one off" commands where it's unlikely that their invocation has been scripted.

pauloscustodio, 2019-02-13:
Agreed. Let's leave it as z80asm.
I started renaming, but it's in two many places...

suborb, 2019-02-18:
I had the same issue with app make.

I'll think I'll close this one since the main clashing culprit has been sorted.

closed 2019-02-18 by suborb

pauloscustodio, 2021-08-03:
@suborb, I think we should rename also z80asm to z88dk-z80asm, to have a uniform naming for all the tools. I had to change a ton of z80asm test files and some Makefiles, but the work is ready in the issue1104 branch.
What do you think? Shall I merge it to master?

suborb, 2021-08-03:
I left this so that in bin there was z80asm and z88dk-z80asm - internally zcc uses z80asm and I think I changed everything outside of src/z80asm to use z88dk-z80asm?

The only reason not to change is compatibility where people are using z80asm directly and invoking z80asm.

pauloscustodio, 2021-08-03:
There was one z80asm in the examples directory.

I can make the z80asm install create a z80asm copy and the external interface is kept, but everything in z88dk uses now z88dk-z80asm.

suborb, 2021-08-03:
That's a good compromise, we can remove the copy at some point in the future.

pauloscustodio, 2021-08-03:
Done.