z88dk/z88dk #1106: (classic) problem using maths with sdcc

zx70, 2019-02-12:
I may be wrong but I thought to be able to build and run simple maths examples also in the classic environment by simply setting the '-compiler' flag to sdcc.

rndsphere.c is now givint the following message:

line 725: symbol '_sqrt' not defined
line 730: symbol '___fs2slong_callee' not defined
line 804: symbol '___slong2fs_callee' not defined
line 813: symbol '___fsadd_callee' not defined
line 822: symbol '___fsdiv_callee' not defined
line 825: symbol '___fs2slong_callee' not defined

My local environment is actually a bit a mockup, thus it can also be my fault.   ..does anybody get the same error ?

suborb, 2019-02-12:
You have to give -lmath48 as well

zx70, 2019-02-12:
works  :)

closed 2019-02-12 by zx70