z88dk/z88dk #1107: (sccz80) Various issues with function pointer calls

suborb, 2019-02-13:
Mostly related to reintroducing the __z88dk_fastcall semantics a few issues back.

So do a cleanup in the area and verify that various combinations work.

suborb, 2019-02-13:
Hopefully resolved and a bunch of tests added (the test file can actually be run with ticks)

pauloscustodio, 2019-02-15:
I'm getting test failures in Issue_1105_function_pointer_calls.c in MINGW64_NT-10.0

```
zcc +test -vn -a Issue_1105_function_pointer_calls.c -o Issue_1105_function_pointer_calls.opt
diff -w tmp.opt results/Issue_1105_function_pointer_calls.opt
53,55c53,58
<       ex      (sp),hl
<       call    l_jphl
<       pop     bc
---
>       pop     af
>       ld      bc,i_2
>       push    bc
>       push    af
>       ret
> .i_2
198c201
<       ld      hl,i_2+0
---
>       ld      hl,i_3+0
204c207
<       ld      hl,i_3+0
---
>       ld      hl,i_4+0
265,269c268,273
<       ex      de,hl
<       ex      (sp),hl
<       push    de
<       call    l_jphl
<       pop     bc
---
>       pop     af
>       ld      bc,i_5
>       push    bc
>       push    af
>       ret
> .i_5
273a278,279
>       push    bc
>       push    bc
309,311c315,320
<       ex      (sp),hl
<       call    l_jphl
<       pop     bc
---
>       pop     af
>       ld      bc,i_6
>       push    bc
>       push    af
>       ret
> .i_6
380c389
< .i_2
---
> .i_3
383c392
< .i_3
---
> .i_4
make[1]: *** [Makefile:28: Issue_1105_function_pointer_calls.opt] Error 1
make[1]: Leaving directory '/D/Data/git/z88dk/testsuite'
```

suborb, 2019-02-18:
@pauloscustodio I thought I left a comment before the weekend but it appears to have vanished into the void.

I've just validated the nightly using wine and I'm not seeing these differences - I think it might be an environmental thing for you?

pauloscustodio, 2019-02-18:
@suborb, yes I suspect mingw might be the cause - I'll check further.

pauloscustodio, 2019-02-19:
@suborb, the issue I had is gone after a clean install of mingw.

closed 2019-02-19 by pauloscustodio