z88dk/z88dk #1116: sccz80 generates wrong code in msys2 ...

pauloscustodio, 2019-03-03:
... and most tests in testsuite fail.

The cause is a broken implementation of tmpfile() in msys2: it creates the temporary file in the root directory of C:\ and therefore needs elevated privileges.
It returns NULL and sets errno to EACCES when run with normal privileges.

This causes the sccz80 compiler to generate wrong code.

pauloscustodio, 2019-03-03:
Note: tmpfile() of MinGW seems OK, it's only in msys2 that it tries to use C:\

suborb, 2019-03-03:
Ouch. Just to confirm: the nightly builds were correct and it was just your environment that was broken?

closed 2019-03-03 by pauloscustodio

pauloscustodio, 2019-03-03:
> Ouch. Just to confirm: the nightly builds were correct and it was just your environment that was broken?

Yes, the nightly builds are built with mingw32, which has a good implementation of tmpfile(), using /tmp.

msys2, which is a free implementation of cygwin, has a bad tmpfile().

The reason I had so strange behavior in my environment is that sometimes I was using sccz80 built by mingw32 in a msys2 environment, and all was well.

In a wrong build, all the stack setup and stack cleanup were missing.

commit 585a7dd9 (2019-03-03): Fix #1116: sccz80 generates wrong code in msys2

Implement my_tmpfile() and use it in win32 builds.
commit e01e197e (2019-03-03): Merge pull request #1117 from z88dk/msys2_fix

Fix #1116: sccz80 generates wrong code in msys2