z88dk/z88dk #1119: (classic) Create uPD7220 video driver

suborb, 2019-03-06:
Used by:

* A5105
* QX10
* NCR DMV
* MZ3500

Datasheet: http://www.vintagecomputer.net/fjkraan/comp/qx10/doc/nec7220.pdf
CP/M driver: http://www.cpm.z80.de/roche/NCRGRAF.ASM
Design manual: http://bitsavers.trailing-edge.com/components/nec/uPD72220_GDC_Design_Manual_Version_3.pdf

suborb, 2019-03-07:
Some progress (a5105/BIC), character set:

[image]

And the usual gencon screen (after applying some guesses):

[image]

zx70, 2019-03-08:
not bad really  !

suborb, 2019-10-21:
Come back onto this one now, source code for the a5105 can be found here: https://hc-ddr.hucki.net/wiki/doku.php/homecomputer/bic

suborb, 2021-05-20:
I didn't enjoy doing this at the time, so it's unlikely it'll ever be finished. Closing so not to raise false hope.

closed 2021-05-20 by suborb

zx70, 2024-10-04:
I put the original qx10 bios sources in docs/targets/epson_qx10