z88dk/z88dk #1120: (classic) Excalibur 64

zx70, 2019-03-12:
Summary:
* MC6845 
* internal timer for sound frequency generation
* Floppy drive
* KansasCity cassette interface
* high resolution graphics
* memory banks, the video memory is in the way when enabled
* udg

Machine boots into BASIC or CP/M, a shortcut permits to use the ROM basic to run programs loaded from within the CP/M.

Potential features:

- [ ] appmake creates a bootable disc
- [ ] gencon text 40 column mode
- [ ] gencon text 80 column mode
- [ ] gencon text scrolling
- [ ] inkey keyboard handling
- [ ] tape / wav support
- [ ] hires graphics support

suborb, 2019-03-12:
That looks like - any idea where to get the CP/M boot disc from since the location mentioned here: https://www.microbee-mspp.org.au/forum/viewtopic.php?f=31&t=1306 does not seem to exist.

The hires mode is implemented in an interesting way, allowing a combining of hires and text at the same time.

zx70, 2019-03-13:
I found few raw disk and cassette files in the MSPP repository.  The config for cpmtools (already present in their modified version):
# Excalibur 80x5x1024 (format 4)
diskdef excalibur4
  seclen 1024
  cylinders 80
  sectrk 5
  heads 2
  sideoffs 128
  blocksize 2048
  maxdir 128
  skew 3
  skewstart 1
  datasect 1
  boottrk 2
  os 2.2
end

suborb, 2019-03-13:
Thanks - I'm far too lazy to write a post there.

Anyway, +cpm -subtype=excali64 will now create CP/M discs so we've got access to the machine.

zx70, 2019-03-13:
perfect,  many thanks !  I'm slowly proceeding with the console support

suborb, 2019-03-13:
Cool, it looks like we need to use the same code_graphics/HIMEM trick that was used on the tiki100 - have the graphics routines located > 32768 and then flip the paging bit to bring in VRAM.

```
in a,($50)
set 0,a
out ($70),a
```

Should do the trick to page in. The console then should be pretty easy - it looks like char data starts at $2000, and then attributes at $2800.

BTW it looks like basck crashes on the 8k rom file.

zx70, 2019-03-13:
yes,  it crashes.   The reason is the  pages folded in the ROM chips confusing basck. It tries to decode the BASIC tables bur they're not 'paged in' correctly when it gets to that address.
I've put in the techdocs repository a partial disassembly and the "techdata.txt" file that I recovered from the disk image and converted.

zx70, 2019-03-13:
if you are at a good point with graphics,  I'm not stopping you  ;)

Ottieni Outlook per Android<https://aka.ms/ghei36>

________________________________
From: Stefano Bodrato
Sent: Wednesday, March 13, 2019 10:21:34 PM
To: z88dk/z88dk
Subject: Re: [z88dk/z88dk] (classic) Excalibur 64 (#1120)

yes,  it crashes.   The reason is the  pages folded in the ROM chips confusing basck. It tries to decode the BASIC tables bur they're not 'paged in' correctly when it gets to that address.
I've put in the techdocs repository a partial disassembly and the "techdata.txt" file that I recovered from the disk image and converted.

suborb, 2019-03-13:
I think I'm just procrastinating about the upd7220 work (still battling screen mode switches). But here's a couple of screenshots (in the feature/excalibur64 branch) - some colour work is needed....

[image]
[image]

zx70, 2019-03-14:
Manually moving the $2000 bottom bytes in ROM1 at position $2000 helps basck to complete a partial analysis:

# File size: 16384

# Specific Z80 CPU code detected

# Microsoft 8080/Z80 BASIC found
#  Extended syntax detected (classic version)
#  Double precision maths detected
#  Microsoft signature not found

BASTXT 	= $FC58   ; BASIC program start ptr (aka TXTTAB)

#    CPDEHL (compare DE and HL), code found at $2B79

INPORT 	= $FC48   ; Current port for 'INP' function
OTPORT 	= $FC4B   ; Current port for 'OUT' statement
TMSTPT 	= $FDD3   ; Temporary string pool pointer
TMPSTR 	= $FDF3   ; Temporary string
NXTOPR 	= $FE13   ; Address ptr to next operator
CURPOS 	= $FDC6   ; Character position on line (TTYPOS on Ext. Basic)
PROGND 	= $FE19   ; BASIC program end ptr (aka VARTAB)
VAREND 	= $FE1B   ; End of variables
ARREND 	= $FE1D   ; End of arrays (lowest free mem)
SGNRES 	= $FE45   ; Sign of result
DIMFLG 	= $FDCE   ; (word), aka LCRFLG (Locate/Create and Type
SAVTXT 	= $FE06   ; (word), prg pointer for resume
TEMPST 	= $FDD5   ; (word), temporary descriptors
TEMPPT 	= $FDD3   ; (word), start of free area of temporary descriptor
CURLIN 	= $FC56   ; (word), line number being interpreted
OLDLIN 	= $FE15   ; (word), old line number set up ^C ...
SAVTXT 	= $FE06   ; (word), prg pointer for resume
OLDTXT 	= $FE17   ; (word), prg pointer for CONT

HALFPI 	= $23F5   ; Half PI constant ptr
HALF 	= $21EA   ; Constant ptr for 0.5 in FP
UNITY 	= $1662   ; Constant ptr for number 1 in FP
LOGTAB 	= $1666   ; Table used by LOG

FPREG 	= $FE41   ; Floating Point Register (FACCU, FACLOW on Ext. BASIC)
LAST_FPREG 	= $FE43   ; Last byte in Single Precision FP Register (+sign bit)
FPEXP 	= $FE44   ; Floating Point Exponent
DBL_FPREG 	= $FE41   ; Double Precision Floating Point Register (aka FACLOW)
DBL_LAST_FPREG 	= $FE4D   ; Last byte in Double Precision FP register (+sign bit)

CPDEHL 	= $2B7A   ; compare DE and HL (aka DCOMPR)
FNDNUM 	= $3A0F   ; Load 'A' with the next number in BASIC program
GETINT 	= $3A10   ; Get a number to 'A'
GETWORD 	= $39F6   ; Get a number to DE (0..65535)
DEPINT 	= $39F9   ; Get integer variable to DE, error if negative
FPSINT 	= $39F5   ; Get subscript
SBSCPT 	= $35DD   ; Sort out subscript
SCPTLP 	= $35E3   ; SBSCPT loop
GETVAR 	= $3501   ; Get variable address to DE
CHKSTK 	= $284D   ; Check for C levels of stack
OPRND 	= $3393   ; Get next expression value
SYNCHR 	= $0008   ; Check syntax, 1 byte follows to be compared
LFRGNM 	= $39D3   ; number in program listing and check for ending ')'
HLPASS 	= $193C   ; Get back from function passing an INT value HL
MIDNUM 	= $39D6   ; Get number in program listing
INT_RESULT_A 	= $17F7   ; Get back from function, result in A (signed)
INT_RESULT_HL 	= $1904   ; Get back from function, result in HL
UNSIGNED_RES_A 	= $36EC   ; Get back from function, result in A

GETYPR 	= $0020   ; Test number FAC type (Precision mode, etc..)
TSTSGN 	= $17BF   ; Test sign of FPREG
_TSTSGN 	= $17FE   ; Test sign in number
INVSGN 	= $17EC   ; Invert number sign
STAKFP 	= $180E   ; Put FP value on stack
NEGAFT 	= $224C   ; Negate number
[... 204 lines cut ...]
$2412 - [227] TAN
$2427 - [228] ATN
$3BA8 - [229] PEEK
$3F14 - [230] VPEEK
$FE78 - [231] PEN
$3F1C - [232] WPK
$FE81 - [233] EOF
$FE84 - [234] JSK
$3F2C - [235] JUMP
$3F32 - [236] DEG
$3F18 - [237] MPK$
$3F36 - [238] RAD
$18E9 - [239] CINT
$191B - [240] CSNG
$1945 - [241] CDBL
$1990 - [242] FIX
$38F7 - [243] LEN
$372A - [244] STR$
$39B9 - [245] VAL
$3903 - [246] ASC
$3913 - [247] CHR$
$3955 - [248] LEFT$
$3985 - [249] RIGHT$
$398E - [250] MID$
$4EC5 - [251] '

zx70, 2019-03-14:
The color attributes include the 'PCGEN' flag (= UDG font) in bit 0, it could be part of the problem shown in your screenshot.

zx70, 2019-03-14:
Another thing to take care of is the video glitches caused by video memory contention.
I'm not sure whether the emulators will show the problem, but the VSYNC signal should be checked before writing to the video page:

    cswait:
      in      a,(50h)        ; get system status
      and     10h
      jr      z,cswait        ; wait for CSYNC - Composite video sync. signal

suborb, 2019-03-15:
Thanks, the colour mistake was because I rotated right instead of left so was setting the bit! They still need mapping from the ANSI colours to the native colours.

I've added in the wait, it does't make a difference on mame, but possibly on the real hardware?

zx70, 2019-03-15:
Probably there is a difference.   It could also be that the problem was solved in hardware in the later models or with extra circuitry (in an article this intention was announced), but the wait loop is still present everywhere in the later ROM.

suborb, 2019-03-16:
If I'm reading mame correctly, it looks like there's 6 x 256 PCG characters available. Each 8x12, padded out to 16 byte boundaries.

For text modes,  I'm taking over PCG bank 0 for a custom font and UDGs. 

However, by combining some of the banks we could support the lores screen (1000 characters) and do pixel addressing to give us a resolution of 320x300

zx70, 2019-03-16:
very nice,  I'm sorry I've been late in disassembling the rom for the PCGEN instruction part.    are the 2 files in techdocs of any use?

suborb, 2019-03-16:
No worries. Combined with the emulator source code it all makes sense. 

Apart from why so much vram was seemingly wasted. 

The next commit should be lores graphics

zx70, 2019-03-25:
I'm testing various ways to merge the compiled program into a .BAS file.
Succeeding with this should provide an elegant way to get three results at once: the short loader (run.com) to pick it from a CP/M disk image, a mockup of the BAS file and the tricks used by 'run.com' to get a sort of .COM file for the Excalibur only, a native format for directly loading it from ROM basic.
All of them would load at 16384.
ATM I don't know where a self booting program should stay.

suborb, 2019-03-25:
It looks like the bootstrap sector(s) are loaded to $d400 - the code to do it as at $0f86 in the ROM.

I'm guessing to make a native, full memory program we'd need to load a short bootstrap which then loads the rest of the code to address $0000

I'm not too certain we'd gain much though given the CP/M TPA is moderately large and provides us with file access etc

zx70, 2019-03-29:
Agreed.  I'd rather try to have support for a configuration with the ROM BASIC paged in (no CP/M at all or the RUN.COM trick).
No luck attempting to append the compiled code directly to the BASIC block; repeated zeros seem to make the BASIC interpreter trim the program and destroy the payload.
I'll try with the classic loader and program data separation.

zx70, 2019-03-29:
Saving Machine Language Programmes on Cassette (found in "Excalibur NEWS #5", Jan '85)
1. Load programme into memory (using monitor programme by L. Adney)
2. Press Reset.
3. Place cassette in recorder.
4. Type CSAVE M "filename", start address, end address
5. Press RETURN. computer responds with READY when file is saved
(for addresses >32767, address=address-65536)

To launch a M/C program (found elsewhere in the newsletters):
POKE -958,LSB
POKE -957,MSB
U=USR(0)

zx70, 2019-04-05:
I'm proud of the appmake results!   CLOADM on MAME works well enough to find the expected bytes in the expected position.    Sadly I couldn't get any kind of output when entering the USR() function, yet.  I'm working on it. 
I created this new converter as an extra option of the Sorcerer.   I'm not sure it was a good idea,  but it works,  even if at the moment only the default bps speed is correctly tuned.

zx70, 2019-04-05:
I was able to get a simple confirmation by passing a return code from within a C program and printing it with PRINT USR(0).
So the skeleton sould be ready, sorcerer.cfg and sorcerer_crt0  :)
@suborb, would you mind to play a little bit more around it ?  The current location I've chosen for the program block is $4100, in the BASIC program area.   I'll be busy for the next few days, so I must stay quiet.

suborb, 2019-04-05:
@zx70 you not checked in the .lst and makefile changes for excalibur_clib.lib

zx70, 2019-04-06:
The reason is that I haven't it yet.   The library I was using was mostly useless and not working,  but the appmake tool now produces a valid wav file for mame which call be loaded and run with:
CLOADM
POKE -958,0
POKE -957,65
PRINT USR(0)

zx70, 2019-04-10:
back.   I'm sorting it out now, I see you chose "excali64" for the target name, and I still can easily adjust it 
;)

suborb, 2019-04-10:
Sorry, I didn't get a chance to do anything - I spent the weekend upgrading computers and other offline things.

Yeah, I'm trying to match the mame name where possible.

zx70, 2019-04-11:
Hope I set up a cleaner skeleton now.
Still the ROM calls seem to have no effect, I'm investigating but I'm clueless (wrong ROM page status?  bad stack area?)..    nevertheless PRINT USR(0) shows the valid return code, so I know something is actually running.
The graphics have something to be fixed too:
'sinwave_c': symbol 'conio_map_colour' already defined in module 'conio_map_colour_noop'

zx70, 2019-04-13:
testing proceeds slowly,  I borrowed from the rom the code used to display 2 asterisks on the top right corner while loading data.   something is shown but wrong data. 
I doubt such effect can be consequence of a wrong conversion in appmake.

zx70, 2019-04-16:
still testing,  the libraries look good but do not work when linked in. 
yet they sometimes work when pasted directly in the crt0 stub!  I suspect something happening when big data chunks are being loaded

zx70, 2019-04-17:
Appmake seems good, I tested it a little more extensively with MAME.    I'm not supporting The 600bps mode but the "--fast" option and the 300 baud mode seem to be in working order,
To switch the excalibur to 300bps:
    POKE -1053,8

(for the record, 1=1200bps (default), 2=600bps).
I think that poking a bigger values would furtherly slow down the LOAD/SAVE routine (e.g.  16 to get to 150bps).

I also think that the tape output bit can be used for bit banging:
    ld      a,0eh			; cassette output signal=low
    out     (63h),a			; Parallel Port Interface Control word

    ld      a,0fh			; cassette output signal=high
    out     (63h),a			; Parallel Port Interface Control word

zx70, 2019-07-05:
@suborb, please don't consider the full support of this target as a requirement for z88dk 2.0, I'm afraid that for loading from BASIC I hit once more a long lasting bug like the one on the X07, NC100 etc..

suborb, 2021-05-22:
I don't think hires is going to get done so closing.

closed 2021-05-22 by suborb