z88dk/z88dk #1122: missing destination file operand after

soydemadrid, 2019-03-16:
Hi I'm trying to build the latest release of z88dk on ubuntu. I'm following your wiki instructions but when I get down to make install I get hundreds of these sorts of errors:

> Try 'install --help' for more information.
> install: missing destination file operand after '{,//usr/local/share/z88dk/}libsrc/cpm/cpm_platform_init.asm'
> Try 'install --help' for more information.
> install: missing destination file operand after '{,//usr/local/share/z88dk/}libsrc/cpm/a_border.asm'
> Try 'install --help' for more information.
> install: missing destination file operand after '{,//usr/local/share/z88dk/}libsrc/cpm/a_machine.asm'
> Try 'install --help' for more information.
> install: missing destination file operand after '{,//usr/local/share/z88dk/}libsrc/cpm/a_ink.asm'
> Try 'install --help' for more information.
> install: missing destination file operand after '{,//usr/local/share/z88dk/}libsrc/gfxosborne1.lst'
> Try 'install --help' for more information.
> 

Thanks for any help

suborb, 2019-03-16:
It looks like it's a dash vs bash difference. Although it's /bin/sh that's being invoked one of them is non-standard and either accepting the substitution syntax or not accepting the syntax. 

Relinking /bin/sh to bash makes it work.

soydemadrid, 2019-03-21:
Thanks I should of seen this in the instructions!! Sorry... Thanks again for the help

closed 2019-03-21 by soydemadrid

suborb, 2019-03-21:
No worries - I only added it to the instructions after you raised the issue!