z88dk/z88dk #1124: Questions about TMS9918A library wrapper for RC2014 target

jblang, 2019-03-19:
@feilipu has been helping me write a wrapper for my TMS9918A assembly language library for the RC2014 target he developed.  I've run into some problems and he suggested I open an issue here to get your input.

I had previously written a simple wrapper that worked with the z88dk +cpm target (using classic lib).  I have the original tms.asm library, tmsc.asm wrapper, tms.h header, and a working example program, hopalong.c in my [github repo](https://github.com/jblang/TMS9918A/tree/master/examples).  Phillip has forked my repo and created a [z88dk-lib wrapper](https://github.com/feilipu/TMS9918A/tree/master/tms) for my code.

There's a [discussion](https://groups.google.com/forum/?utm_source=digest&utm_medium=email#!topic/rc2014-z80/ffORM4ZQ7xE) on the RC2014 mailing list for further context.

This first issue is specifically what Phillip suggested I ask about, but I thought I'd run these other issues by you as well....

1) newlib seems to be missing "conio.h" which I'm using for the kbhit() and getch() functions.  For now I have just commented the calls out and made my program loop indefinitely, but it ideally needs some non-blocking way to check for keyboard input.  

2) Trying to compile the example using this invocation, and the -o option is not honored:
zcc +rc2014 -subtype=cpm -lm -llib/rc2014/tms -create-app -ohopalong hopalong.c

I get hopalong_CODE.bin, hopalong_BSS.bin, hopalong_UNASSIGNED.bin, and hopalong.ihx, but no com file. It does work when using the cpm target:

zcc +cpm -lm -create-app -ohopalong hopalong.c tmsc.asm

Phillip suggested I can just copy hopalong_CODE.bin to hopalong.com, and that's what I've done.

3) The program starts to execute but then hangs when it reaches the first %f format specifier in a printf call. Everything up to that point is output correctly on the terminal.

4) If I comment out the printf, the program activates the TMS9918 display but the output is corrupted. I'm seeing some red vertical bars with black dots but nothing resembling the expected output. The same code works when I compile it for the cpm target using the invocation above.  I haven't spent any time debugging this yet, but I suspect something is wrong with the floating point since I also had the aforementioned problem with the printf %f specifier.

feilipu, 2019-03-19:
I’m not 100% certain I didn’t mess up the wrapper or library, but the float %f should work, and the .COM appmake output should work too.

Perhaps we start with a simple example to test and resolve, before the complexity of z88dk-lib and the actual library code?

feilipu, 2019-03-23:
I've written a small C program, to test that the floating point format converter works on hardware.
```c
#include <stdlib.h>
#include <stdio.h>

// zcc +rc2014 -subtype=cpm -v --list -m -SO3  --allow-unsafe-read --c-code-in-asm --opt-code-size -lm --max-allocs-per-node200000 @ftest.lst -o ftest
// zcc +yaz180 -subtype=cpm -v --list -m -SO3 --allow-unsafe-read --c-code-in-asm --opt-code-size -lm --max-allocs-per-node200000 @ftest.lst -o ftest

#pragma printf = "%s %c %u %f"     // enables %s, %c, %u, %f only 

void main(void)
{
    printf("float: %f\r\n", (float)3);
}
```
This works as expected, when running the resulting `ftest_CODE.bin` file as `FLOAT.COM`. Probably the only note is that (I think) the `%f` conversion is turned off normally to save space, so it does need to be enabled by the `#pragma`.

`ftest.lst` contains one line referring to `ftest.c`. Further files could be added to the `.lst` file on separate lines.
```
>a:xmodem float.com /r

File created
Receiving via CON with CRCsCC
B>float

float: 3.000000

B>
```
I will have a look at the other issues now too.

feilipu, 2022-02-08:
Closing, aged out.

closed 2022-02-08 by feilipu