z88dk/z88dk #1129: [sdcc][overflow] unsigned char overflows with 254

Fabrizio-Caruso, 2019-04-05:
If I compile the pv1000 target in Cross Chase with sdcc I get an overflow warning at
```
 for(k=0;k<254;++k){};
```

The compilation produces:
```
/cygdrive/c/z88dk/bin/zcc.exe +pv1000 \
-compiler=sdcc \
...
-lndos \
-clib=default -pragma-redirect:fputc_cons=fputc_cons_generic \
...
-create-app -o../build/FULL_pv1000.bin
chase/level.c:171: warning 158: overflow in implicit constant conversion
chase/main.c:658: warning 126: unreachable code
...
```

suborb, 2019-04-05:
What type is k?

Fabrizio-Caruso, 2019-04-05:
k is an unsigned char:

void _spiral_slow_down()
{
        unsigned char k;

        for(k=0;k<254;++k){};
}

suborb, 2019-04-05:
That looks like zdcc is applying range changing for a signed char value rather than an unsigned char value.

The code generated looks right in this case though

zx70, 2019-04-06:
let's explicitly alert @spth   ;)

spth, 2019-04-06:
This issue might be related to a known bug (https://sourceforge.net/p/sdcc/bugs/2733/), but I'm not sure, so I'd recommend filing a new bug report at https://sourceforge.net/p/sdcc/bugs/.

However, since this is not a particularly bad issue (after all the generated code is correct), it most likely won't be fixed before the 3.9.0 release in April.

feilipu, 2019-07-04:
Is this still valid, now that r11311 zsdcc is in place?

Fabrizio-Caruso, 2019-07-08:
@felipu, @spth, I confirm the problem is still them even in the latest ZSDCC version as of today (8th July 2019)

spth, 2019-07-09:
This is probably one of these upstream bugs:
https://sourceforge.net/p/sdcc/bugs/2733/
https://sourceforge.net/p/sdcc/bugs/2877/

Fabrizio-Caruso, 2019-09-15:
Bump

This issue is still there with the latest snapshot as of 9th Sept 2019:
...
chase/level.c:171: warning 158: overflow in implicit constant conversion
chase/main.c:681: warning 126: unreachable code
...

feilipu, 2019-09-16:
Both of these issues are still open on the upstream sdcc bug tracker.
You can subscribe to both tickets to see when they're fixed.

feilipu, 2020-01-22:
Both the upstream sdcc bugs remain open.
- [2733](https://sourceforge.net/p/sdcc/bugs/2733/)
- [2877](https://sourceforge.net/p/sdcc/bugs/2877/)

feilipu, 2026-10-14:
Checked on zsdcc 4.6.0 r16639.

I can't see SDCC fixing these two outstanding bugs anytime soon.
So since the code is right, we just look the other way.

```c
unsigned char k;

for (k = 0; k < 254; ++k) {}
```

The compiler still prints `warning 158: overflow in implicit constant conversion`. The Z80 code is a 254-iteration loop:

```asm
ld b,#0xfe
djnz loop
```

The object code is correct. The warning remains. The same warning also appears for `unsigned char c = 1 << 7` and for `192 + 41` passed as an `unsigned char`.

This front end is SDCC. The z88dk patch does not change it. These upstream reports are still open:

- https://sourceforge.net/p/sdcc/bugs/2733/
- https://sourceforge.net/p/sdcc/bugs/2877/

closed 2026-10-14 by feilipu