z88dk/z88dk #1135: (fp) Create Microsoft single precision maths library

suborb, 2019-04-15:
The single precision library is 32 bit so should fit in nicely with the new maths functionality in sccz80.

Should be able to extract it from the NASCOM rom here: http://www.nascomhomepage.com/lang/8kbasic.asm

zx70, 2019-04-16:
that's more challenging than as it looks, especially to port,  but yes..  nice to have

suborb, 2019-04-18:
I managed to mostly separate it out and it's broadly functioning.

However, the performance is pretty awful (down about 20%) compared to both genmath and math48 for the arithmetic operations.

Because the code is written for overall package size, the separation isn't wonderful so the size difference isn't significant either.

As a result, I can't see any reasons to pull in the library directly. However, I think it's worthwhile to keep the bindings (and add some more) to support those machines with single precision in the ROM - that way we'll get the size savings so there is an advantage.

zx70, 2019-04-18:
I think it's great to have the bindings, I'd really like to find some use for basck  :)

suborb, 2019-04-18:
I'm binding to the Laser500 ROM at the moment - basck has identified most of the entry points, or at least roughly in the right area.

I've been through a few ROMs and it looks like most of them use the 40 bit variant rather than the 32 bit: and the Laser500 seems to have some 64bit routines as well. I've added generation of doubles in those formats to sccz80 so I'll see if I can do a similar trick there as well.

zx70, 2019-04-19:
wow you are doing GREAT!

suborb, 2019-04-20:
32 bit is in PR #1137 - I'm sure there's some constants that I've not redefined properly. However it's good enough for normal usage.

closed 2019-04-20 by suborb