z88dk/z88dk #1139: Docker doesn't build

mbarchein, 2019-04-21:
Build error:

```cc -g -O2 -Wall -std=gnu11 -MMD -I../../ext/optparse -I../../ext/regex -o z88dk-basck basck.o 
install z88dk-basck /opt/z88dk/bin/z88dk-basck
make[1]: Leaving directory '/opt/z88dk/support/basck'
make -C testsuite
make[1]: Entering directory '/opt/z88dk/testsuite'
zcc +test -vn -a long_addition_opts.c -o tmp1.opt
diff -w tmp2.opt results/long_addition_opts.opt
mv -f tmp1.opt long_addition_opts.opt
zcc +test -vn -a Issue_482_optrule.c -o tmp1.opt
diff -w tmp2.opt results/Issue_482_optrule.opt
mv -f tmp1.opt Issue_482_optrule.opt
zcc +test -vn -a Issue_600_fastcall.c -o tmp1.opt
sccz80:"Issue_600_fastcall.c" L:6 Warning:Cannot define function 'func4' as __z88dk_fastcall __stdc when it has more than 1 argument [-Wincorrect-function-declspec]
diff -w tmp2.opt results/Issue_600_fastcall.opt
mv -f tmp1.opt Issue_600_fastcall.opt
zcc +test -vn -a Issue_188_builtins.c -o tmp1.opt
diff -w tmp2.opt results/Issue_188_builtins.opt
mv -f tmp1.opt Issue_188_builtins.opt
zcc +test -vn -a Issue_1016_double_const_compare.c -o tmp1.opt
make[1]: *** [Makefile:36: Issue_1016_double_const_compare.opt] Error 1
make[1]: Leaving directory '/opt/z88dk/testsuite'
make: *** [Makefile:168: testsuite] Error 2
The command '/bin/sh -c cd ${Z88DK_PATH}     && chmod 777 build.sh     && ./build.sh     && svn checkout -r 10892 svn://svn.code.sf.net/p/sdcc/code/trunk/sdcc ${SDCC_PATH}     && cd ${SDCC_PATH}     && patch -p0 < ${Z88DK_PATH}/src/zsdcc/sdcc-z88dk.patch     && ./configure --disable-mcs51-port --disable-gbz80-port --disable-avr-port --disable-ds390-port                    --disable-ds400-port --disable-hc08-port --disable-pic-port --disable-pic14-port                    --disable-pic16-port --disable-stm8-port --disable-tlcs90-port --disable-s08-port                    --disable-ucsim --disable-device-lib --disable-packihx     && make     && mv ./bin/sdcc ${Z88DK_PATH}/bin/zsdcc     && mv ./bin/sdcpp ${Z88DK_PATH}/bin/zsdcpp     && cd /     && rm -fR ${SDCC_PATH}     && apk del .build_deps' returned a non-zero code: 2
```

Tried to build with `docker build -t z88dk -f z88dk.Dockerfile .`

mbarchein, 2019-04-21:
[build-error.txt](https://github.com/z88dk/z88dk/files/3101588/build-error.txt)

closed 2019-04-21 by suborb

suborb, 2019-04-21:
Sorry, all sorted now: an issue with the new floating point code.

mbarchein, 2019-04-21:
Thanks, that was fast! Now it's compiling fine

commit b1f1befe (2019-04-21): Fix #1139: Incorrect buffer size/stack overflow fun