z88dk/z88dk #1145: (build) Intermittent build failures

suborb, 2019-05-08:
Started fairly recently - travis infrastructure upgrade?

* +zx xample.c 
* +osca wc.c
* Random macOS assembling .c files (not seen in a while)

Seem to be the culprits. Running under valgrind here shows nothing unusual.

suborb, 2019-05-17:
150 runs of +osca wc.c under valgrind showed nothing with sccz80 on Ubuntu 18.10

A few (random) z80asm errors though (2/190):

```
Warning at file 'wc.c' line 512: integer '65562' out of range
Warning at file 'wc.c' line 512: integer '1703936' out of range
```

Unfortunately I don't have the .asm file generated.

suborb, 2019-08-10:
We're getting intermittent test failures on: Issue_1132_struct_parameter.c - similar issue to above.

I've put some more error trapping in both sccz80 and copt to see where the issue is coming from but of course it's now not failing!

suborb, 2019-08-11:
So the failure is due to the joining together of two lots of a stack cleanup:

```
        exx
        ld      hl,16   ;const
        add     hl,sp
        ld      sp,hl
        exx
        ld      hl,14   ;const
        add     hl,sp
        ld      sp,hl
```

Is turned into:

```
        ld      hl,30   ;const
        add     hl,sp
        ld      sp,hl
```

I've added logging to copt to show the addition.

Looking at the most recent PR we had one failure, which gave this output:

```
zcc +test -vn -a Issue_1132_struct_parameter.c -o tmp1.opt
RPN adding 10 + 4
RPN adding 10 + 4
RPN adding 14 + 16
RPN adding 14 + 14
diff -w tmp2.opt results/Issue_1132_struct_parameter.opt
50c50
<       ld      hl,8    ;const
---
>       ld      hl,30   ;const
72c72
<       ld      hl,8    ;const
---
>       ld      hl,28   ;const
```

After a poke it gave us this:

```
zcc +test -vn -a Issue_1132_struct_parameter.c -o tmp1.opt
RPN adding 10 + 4
RPN adding 10 + 4
RPN adding 14 + 16
RPN adding 14 + 14
diff -w tmp2.opt results/Issue_1132_struct_parameter.opt
zcc +test -vn -a Issue_569_jagged_array.c -o tmp1.opt
```

So the maths is working (phew), but we're getting the wrong result sometimes - I can't reproduce this on any of my machines - I've tried thousands of times on both x86_64 linux, MacOS and a Raspberry Pi.

suborb, 2019-09-13:
Got a good one:

`
zcc +test -vn -a Issue_1132_struct_parameter.c -o tmp1.opt
2065RPN adding 10 + 4 = 14
2066RPN evaluation resulted in: 14
2067RPN adding 10 + 4 = 14
2068RPN evaluation resulted in: 14
2069RPN adding 14 + 16 = 30
2070RPN evaluation resulted in: 5
2071RPN evaluation resulted in: 0
2072RPN adding 14 + 14 = 28
2073RPN evaluation resulted in: 5
2074diff -w tmp2.opt results/Issue_1132_struct_parameter.opt
207550c50
2076< 	ld	hl,5	;const
2077---
2078> 	ld	hl,30	;const
207972c72
2080< 	ld	hl,5	;const
2081---
2082> 	ld	hl,28	;const
```

Almost like something is being left on the RPN stack on it, but given that these are really simply expressions I'm not sure how.

ped7g, 2019-09-13:
any particular c/cpp file responsible for the values? I may review it in "dry" way (just by reading the source on github) if it's sufficiently self-contained for such approach (still didn't get myself to install the z88dk locally and to learn how to use it :/ :) .. sorry)

suborb, 2019-09-13:
Another eye is always going to be handy. Not being able to reproduce this locally and then only intermittently remotely is challenging to say the least.

The calculator is in src/copt/copt.c - the extra logging is in the feature/gbz80 branch

The rule that's causing problems is something like this:

```
%check 0 <= %2 <= 65535
        ld      hl,%1   ;const
        add     hl,sp
        ld      bc,%2
        add     hl,bc
=
        ld      hl,%eval(%1 %2 +)       ;const
        add     hl,sp
```

It's the expression `%eval(%1 %2 +)` that is ending up with the wrong result.

suborb, 2019-09-28:
I think this one is now solved so closing

closed 2019-09-28 by suborb