z88dk/z88dk #1150: (sdcc) Pre-built sdcc.exe claims libgcc_s_seh-1.dll to be missing

zx70, 2019-05-16:
ditto

zx70, 2019-05-16:
My fault, i was using sdcc in place of zsdcc  :(

zx70, 2019-05-16:
nope, the zcc option "compiler=zsdcc" is not a valid workaround (probably falls back to zccz80)

zx70, 2019-05-16:
fixed, local problem  :(

zx70, 2019-05-16:
Still problems, sorry for reiterating the incident, my current env is a mess.

suborb, 2019-05-20:
Hopefully the next build will have the right dependencies in the kit.

Sadly the kit size will be huge since libstdc++ is one of them and it's 15M

zx70, 2019-05-21:
Works !   Thankyou, the ZIPed archive is only 4MB bigger.
My feeling is that the current kit is now more mature, the collider.c demo couldn't be built for quite a while after including sdcc and evolving the whole kit; now it is valid again, thus I put it in the "examples" folder.
In the spare time I'm also adjusting the demos to make them compatible with SDCC (microman is very close).

closed 2019-05-21 by zx70