z88dk/z88dk #1151: Test Failure during build of git nightly on DietPi (Rasbian) & RasPi 3B+

Xalior, 2019-05-17:
To reproduce:
  git clone --recursive https://github.com/z88dk/z88dk.git
  cd z88dk
  chmod 777 build.sh
  ./build.sh

Expected Results:
  A build ;-)

Actual Results:
  zcc +test -vn -a Issue_1105_function_pointer_calls.c -o tmp1.opt
  diff -w tmp2.opt results/Issue_1105_function_pointer_calls.opt
  165,167c165,166
  < 	ld	hl,0	;const
  < 	ld	d,h
  < 	ld	e,l
  ---
  > 	ld	hl,65535	;const
  > 	ld	de,65535
  Makefile:36: recipe for target 'Issue_1105_function_pointer_calls.opt' failed
  make[1]: *** [Issue_1105_function_pointer_calls.opt] Error 1
  make[1]: Leaving directory '/u/GIT/z88dk/testsuite'
  Makefile:168: recipe for target 'testsuite' failed
  make: *** [testsuite] Error 2
  root@maggie:/u/GIT/z88dk# 

Raspberry Pi 3B+ Environment:
  root@maggie:/u/GIT/z88dk# uname -a
  Linux maggie 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

  root@maggie:/u/GIT/z88dk# cat /etc/issue.net 
  Raspbian GNU/Linux 9 (DietPi)

  root@maggie:/u/GIT/z88dk# gcc -v
  Using built-in specs.
  COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/6/lto-wrapper
Target: arm-linux-gnueabihf
  Configured with: ../src/configure -v --with-pkgversion='Raspbian 6.3.0-18+rpi1+deb9u1' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
  Thread model: posix
  gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1+deb9u1) 

Please let me know if you need anything else? :)
  
-Thanks in Advance,
 Dx

suborb, 2019-05-17:
This is really similar to #1020 and I think has the same root cause, just in a different place. 

I'll fix it up this weekend.

suborb, 2019-05-19:
I believe it was fixed in the commit above - certainly it passed on my pi.

If not, please feel free to re-open.

closed 2019-05-19 by suborb

Xalior, 2019-05-21:
Clean build confirmed. Thank you <3 x