z88dk/z88dk #1155: (newlib) Update newlib to support math32

suborb, 2019-05-21:
The classic PR provides support for math32 in classic. However, newlib doesn't fully support math32 yet. The following is my understanding of what needs to be done:

- [ ] Update .lst files to include the math32 core into each of the library builds
- [ ] Generate math32.lib for newlib (sorry, might as well keep the library names aligned)
- [ ] Update printf core to handle 4 byte floats for sccz80
- [ ] Update printf core to not convert 4 byte floats to math48 floats
- [ ] Provide support routines needed for dtoa and hence printf() to work
- [ ] Update scanf core to handle 4 byte floats for sccz80
- [ ] Update scanf core to not convert math48 floats to sdcc floats
- [ ] Create math32 support routines for strtod - these should be in -lmath32
- [ ] Update math.h for math32 values/functions
- [ ] Check for any other float usage in library, ensure it uses asm_ interfaces and that they are present in -lmath32

feilipu, 2019-05-21:
Can't do everything, but will at least try to get the asm pieces working.

In #1157 
- The .lst files for all targets have some build info already, but this will be extended to all math32 asm functions.
- Add in sections specific to math32.
- Build the `-lmath32` asm bindings

.

feilipu, 2019-05-21:
BTW, I'm pretty impressed with the zsdcc generated functions in the classic lib.
I don't think it would  be possible to do a substantially better job by hand, unless one avoided usage of the index registers. But the saving in ticks would be outweighed by the increase in maintenance effort, imho.
Although, still need to work on the coefficients for greater accuracy from the functions. I suspect that the Hi-Tech C coefficients were designed primarily for performance.

feilipu, 2019-05-21:
On the topic of performance optimisation.

The math32 library can be performance optimised easily, simply by managing the number of N-R iterations and/or the number of Horner's coefficients processed.

Is it worth adding in a "FAST&LOOSE" vs "SLOW&ACCURATE" compile switch for the library?

Although slow is a relative term, given that it is still faster (?) than the d48 library even when running slow and accurate.

suborb, 2019-05-22:
The basic arithmetical operations should always be accurate. I can see an argument for reducing the number of coefficients, but we should probably see what effect that actually has in terms of performance.

Since the coefficients are located in separate files, if the number of coefficients was also defined in that file then they could be easily overwritten by a different library/user code.

As you say, the performance before the accuracy changes was significantly faster than any other library so it may be that we don't need to make this trade off.

feilipu, 2019-05-22:
I don't think it will make too much difference by modifying coefficients for the Horner's method, really. The `poly()` function is pretty fast, with simply multiplies and adds. Saving two coefficients is only reducing by 18 `16_8x8` multiplies, for example. In the z80_zxn `mul de` world, this is hardly worth considering.

It is really only the gaming world that might benefit from an inaccurate Quake style normal calculation using inverse square root.

Let's get a definitive view (now there's a bit of stability) to see where things are at in terms of performance before doing anything further.

feilipu, 2019-07-19:
I'm playing with the supporting functions currently, and I find a fairly knotty problem.
The current use of [`_fpclassify`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math32/z80/f32_fpclassify.asm) assumes `dehl`, but the [newlib functions](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/stdlib/z80/__dtoa__.asm) assume the floating point number is in the exx registers, which could be `dehl'`.

I'd rather not (and cannot) rewrite all of the newlib printf variants to expect the float in the main registers, not least because the main registers are filled with parameters. It makes sense to use exx registers for the number (whether it is 48-bits or 32-bits or whatever). But, this means changing the current meaning of `fp_classify`.

Thoughts?

suborb, 2019-07-19:
My thoughts are that asm_XXXX should stick to the ABI that @aralbrec created - the floating point value in the alternate set - that will keep newlib dtoa/printf etc working.

I don't think there's anywhere in classic that calls asm_fpclassify directly - and if it did it can be fixed to respect the ABI.

feilipu, 2019-07-19:
Ok. I'll make a smallish commit to deal with this and its `__dtoa__` friends shortly.

feilipu, 2019-07-22:
So I needed `log10f()` for calculating digits, so I added that and some friends from Ceres.
Big win was finding my [stupid mistake with `expf()`](https://github.com/z88dk/z88dk/pull/1240/commits/79f01a4a5de3f8e1626c665c25d2a359de1686ef?file-filters%5B%5D=.c#diff-4bdf8fd60a07bfee06542515af7ee909L75), which was substantially responsible for any inaccuracy. Now all the log and power functions are pretty good (noting 7 significant digit limit).

But, there's an odd error with the test suite that I can't replicate on my testing. Something going wrong with the power function when the base is [presumably less than 0.7](https://github.com/z88dk/z88dk/pull/1240/commits/79f01a4a5de3f8e1626c665c25d2a359de1686ef?file-filters%5B%5D=.c#diff-01a81b568341f6edba45908d86961824R74). i.e `pow(0.5,2)= 0.25` And, I think this is originated in `logf()`, but currently not sure

feilipu, 2019-07-27:
Added PR #1253
- [x] rationalised and added to constants, zero, infinity, and nan usage.
- [x] removed final references to z80-zxn
- [x] add error handlers
- [x] add alternate register conversions
- [x] add alternate register multiply & add

feilipu, 2019-07-28:
@aralbrec I think the only pieces remaining are to convert printf and scanf functions to not convert to 6 byte floats, and build a switch to enable the IEEE library. Neither of which I am comfortable to do. I would appreciate help, if you've time please.

aralbrec, 2019-07-30:
The printf family should be independent of floating point format as they go through dtoa, etc, to do floating point conversion to text.  They do create space on the stack to hold the ascii conversion result but that space should be sufficient for a lower precision float format.

The dto? functions should also be independent of floating point format and call on primitives in the float library to do lower level functions that deal with a specific format.  They call a math lib function to find out how many significant digits are supported and do not try to generate more digits beyond that while implementing all the format specifiers required by the c standard.  They also print nan, etc, if the float library indicates these sorts of things via fpclassify.

scanf of floats is not supported yet as a state machine is needed to qualify a float string but they will go through strtod eventually.  The workaround is to first scan with %s or %[ and then run strtod or atof on the result.  strtod should also operate in a manner independent of float format.

I will try to have a look soon.  I'm still quite a bit backed up but remind me again if progress is too slow :)

-lmath32 or -lm32 ?

feilipu, 2019-07-30:
The linkage selected for classic is `-lmath32`, so it would be agreeable just to align to that for newlib.

There's a lot of untested helper functions and linkages in place now with #1240, and #1253. So the first thing, imho, is to develop a way to link in the library, so that they can be confirmed to work.

The only known weakenss is that the [comparison](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math32/z80/f32_fscompare_callee.asm) operator needs to recognize `-0`. Currently they only do `+0`. I've got a new version in the works, but I'm not happy with it. I don't have a clear enough understanding, to make the code simple and robust.

suborb, 2019-07-30:
> It would be good if the long -fp-mode=ieee -lmath32 -pragma-define:CLIB_32BIT_FLOAT=1 incantation could be shortened, and perhaps rationalised so that there is (optional) automatic selection of z180 or z80n libraries for their respective targets.

Yes, you can define aliases in the .cfg file for this - take a look at test.cfg and you'll see that you can give --math32 to add in the required magic - particularly useful for classic since there's so many maths libraries to choose from.

> The only known weakness is that the comparison operator needs to recognize -0. Currently it only does +0. I've got a new version in the works, but I'm not happy with it. I don't have a clear enough understanding, to make the code simple and robust.

Sorry about that, figuring out that out cleanly eluded me at the time as well. I was thinking about pulling out the exponents of both sides and doing an explicit 0 check. For an edge case it didn't feel right though.

Back to this issue. I think some of the problem is that some of the maths primitives (eg dload/dmul) ended up in the machine library rather than maths library - I can't remember if we moved them around to get that separation though? I'm think it's okay if the implementation, eg `m32_XXXX` is in the machine library, but the default linkage: `dload/dloadb` needs to be in `math32.lib`

Looking through the newlib and classic printf code there's a different way of dealing with the different sizes on the stack.

Classic does this sort of thing: https://github.com/z88dk/z88dk/blob/master/libsrc/stdio/__printf_handle_f.asm#L56

- checks for whether it's sdcc/sccz80
- if sccz80, checks what size the float is
- pads out arguments everything out on the stack so the rest of the code is common (it calls the `ftoa/ftoe` C binding which with l->r pushing means that we can just call it with space for 8 byte floats)

The sccz80/sdcc branch check doesn't need to be dynamic in newlib - it's an assemble time option.

The float size checking appears to happen here in newlib: https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/stdio/z80/asm_vfprintf_unlocked.asm#L464

i.e. which table the value is in defines how much of a stack offset to apply after doing the `va_arg()` equivalent.

feilipu, 2019-07-30:
> > It would be good if the long -fp-mode=ieee -lmath32 -pragma-define:CLIB_32BIT_FLOAT=1 incantation could be shortened.

> Yes, you can define aliases in the .cfg file for this - take a look at test.cfg and you'll see that you can give --math32 to add in the required magic.

Forgot to note the final piece of magic needed `-D__MATH_MATH32` required to access the include file required for extra intrinsic functions, :spiral_notepad: just noted here so I don't forget. 

>> The only known weakness is that the comparison operator needs to recognise -0.

>  I was thinking about pulling out the exponents of both sides and doing an explicit 0 check. For an edge case it didn't feel right though.

Yes it gets messy quickly. I was thinking that both `+0` and `-0` are defined if the exponent is zero, mantissa doesn't matter, and that is already known after `sla e` `rl d` `jr Z,iamzero` and trying to use that fact. But my code to split the range up from there got ugly, and I wasn't sure my thinking was clear. I think it could still be short, but there would be multiple similar tails to the function.

feilipu, 2019-08-02:
Fixed the `-0` problem in [`fs32_compare_callee`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math32/z80/f32_fscompare_callee.asm) . Both `+0` and `-0` are recognised. Side benefit that comparing anything to 0 is now faster, so C `djnz` loops should run better...

But, uncovered a problem that wasn't found by the testing.  Tests of `N op. -M` always fail. This is not related to the new zero detection code. More investigation needed. Pretty certain that negative numbers are not being handled fully.

To fix our floating point number range so comparisons work correctly, we define the following rules:
- Always flip the sign bit.
- If the sign bit was set, flip all the other bits too.

__EDIT__. [Fixed](https://github.com/z88dk/z88dk/commit/0000affed8aaaff4996c00e57be08654a4c21a31) support for negative numbers in comparisons. Probably the [comparison test suite](https://github.com/z88dk/z88dk/blob/master/test/suites/math/math.c#L12) should include a negative number, to catch this kind of thing. All tests range now passing.

__EDIT__. Added changed the [comparison test](https://github.com/z88dk/z88dk/blob/master/test/suites/math/math.c#L12) to include a negative number.

feilipu, 2019-08-13:
Well that's promising. Decided to add `-lmath32` to a newlib command line just to see what would blow up, and there was only one function missing `dload`. And, now it is no longer missing #1265.

So the math32 newlib library seems to "just work". Of course the conversions for both decimal and hex are wrong, and need fixed, but there's a sign of light at the end of this tunnel.

feilipu, 2019-08-27:
Not quite sure why, but the use of the [-D__MATH_MATH32 define in the config file](https://github.com/z88dk/z88dk/commit/23fbfe2a040ef1ef9c66ffc58c18a710d3664a0d) doesn't seem to work for user applications. It is necessary to add it to the command line to [enable these functions](https://github.com/z88dk/z88dk/commit/3882a276608bfa0558a900d6ef79082f39e9bd78) to be found. 😕

feilipu, 2019-09-01:
@suborb 3 of the 4 combinations of `math32` are working now. The only one that doesn't is the sccz80 newlib version. I'd like to get that option working too.

Working:
```sh
zcc +cpm -v --list -O3 --math32 float.c -o float -create-app
zcc +cpm -compiler=sdcc -v --list -O3 --math32 float.c -o float -create-app
zcc +cpm -clib=sdcc_iy -v --list -O3 --math32 float.c -o float -create-app
```
Not working:
```sh
zcc +cpm -clib=new -v --list -O3 --math32 float.c -o float -create-app
```
I think one of the missing pieces of this is to have a sccz80 version of the C functions, because currently the sccz80 new lib doesn't include all the sdcc helper functions (as the classic lib does).

Why don't I do this?
- I don't have enough knowledge of make to write commands to get a version compiled elegantly and,
- I'm not sure whether the right path is to have a sccz80 version of the C functions or, to include all sdcc helper functions in the sccz80 new lib linking (probably not) as done with classic lib.

Anyway, if you have some to think about this, make a call on the way you think is the right way to do this, and if you could extend the [Makefile](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math32/c/Makefile) to create relevant files, it would save me fumbling about on this. I can finish up once you set the direction.

feilipu, 2019-09-11:
Found a very useful tool to [calculate polynomial coefficients using Remez method](https://github.com/samhocevar/lolremez). Used it to fix the `atan()` accuracy from the woeful HiTech C version. Also faster because there is no divide required. The Whetstone benchmark is now nearly right.

suborb, 2019-09-11:
> I think one of the missing pieces of this is to have a sccz80 version of the C functions, because currently the sccz80 new lib doesn't include all the sdcc helper functions (as the classic lib does).

Sorry @feilipu - I was away at the end of August and completely missed this.

I think my preference is to have the sdcc primitives in the sccz80 newlib library. Recompiling for sccz80 is pretty easy however:

- We'd need to maintain two copies of the stubs/trampolines for functions like `atan2()`.
- Keeping all targets/compilers using the same underlying code feels like the right thing to do
- Makes integration of 3rd party libraries that may use sdcc as part of their build step easier

feilipu, 2019-09-13:
One issue I see with with newlib is that the sccz80 compiler is [expecting](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/sccz80/math.h#L227) / producing fastcall functions, but the [linkage for fastcall](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math32/lm32/c/sccz80/sqrt_fastcall.asm) is not connected to the normal names.

These two excerpts from the same c code.

`zcc +cpm -v --list -m -SO3 --c-code-in-asm --math32 @ftest80.lst -o ftest`
```
102   0093              	C_LINE	103,"./ftest.c"
103   0093  21 1A 00    	ld	hl,26	;const
103   0096  39          	add	hl,sp
103   0097  CD 00 00    	call	l_glong
103   009A  D5          	push	de
103   009B  E5          	push	hl
103   009C  CD 00 00    	call	log
103   009F  C1          	pop	bc
103   00A0  C1          	pop	bc
103   00A1  D5          	push	de
103   00A2  E5          	push	hl
103   00A3  CD 00 00    	call	exp
103   00A6  C1          	pop	bc
103   00A7  C1          	pop	bc
103   00A8  C1          	pop	bc
103   00A9  CD 00 00    	call	l_plong
```

`zcc +rc2014 -subtype=cpm -clib=new -v --list -m -SO3 --c-code-in-asm --math32 @ftest80.lst -o ftest`
```
102   0093              	C_LINE	103,"./ftest.c"
103   0093  21 1A 00    	ld	hl,26	;const
103   0096  39          	add	hl,sp
103   0097  CD 00 00    	call	l_glong
103   009A  CD 00 00    	call	log
103   009D  CD 00 00    	call	exp
103   00A0  C1          	pop	bc
103   00A1  CD 00 00    	call	l_plong
```

Somewhere (probably in the include proto) the [definition made for callee](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/sccz80/math.h#L129) needs to be created to link the fastcall function name to normal function name.

This is done for by M4 for sdcc with the [`__DPROTO()`](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/proto/math.h#L153) definition. But for sccz80 this doesn't happen.

That is, unless I'm missing something?

suborb, 2019-09-14:
That linkage pattern does seem wrong - I think there was another issue about it not so long ago. 

Having fn being fastcall will cause issues with function pointers. 

I wonder if it’s worth doing the same as classic and having a separate math.h for each library?

feilipu, 2019-09-15:
I'd guess the most consistent outcome would be to use the m4 to generate linkage for sccz80/new fastcall, just like it does for callee linkage. This would be then identical to the handling done for sdcc/new fastcall and callee.

But, that would mean that sccz80 would need to be modified to produce `fn_fastcall` calls for normal functions (where it expects to call a fastcall function), and `fn` calls for function pointers, etc. I don't know how hard it would be to achieve this differentiation, and how many other things would be broken along the way.

I've just deleted all the `~/lm32/sccz80/fn_fastcall.asm` files from math32. But, it would probably be the more correct solution to put them all back and have these called by sccz80.

feilipu, 2019-09-17:
@suborb was tracking down why printing in sccz80/newlib is off, and realised that [this](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/stdio/z80/asm_vfprintf_unlocked.asm#L473) hasn't been converted to 4 byte floats. And, then I was reminded by your unticked box at top of this issue.

So, I guess that I shouldn't expect it sccz80/newlib to work at this stage?

Probably then best to hold off until the calling convention above is sorted too.
Awaiting your thoughts.

suborb, 2019-09-17:
Sorry I've been having a bit of fun with the gbz80 stuff lately.

Regarding printf, yes, it won't work without that fix (unless you get incredibly lucky with stack arguments), it should be a fairly smallish insertion to do it, it's a question of finding the appropriate point.

The `__z88dk_fastcall` function pointer calling is really tricky, in order for it to work the function pointer has to be marked as fastcall pointer - this will ensure that `dehl` are preserved - the code generated for this scenario is pretty awful and performs worse than a regular function pointer call. This is required for _math32_ since the parameter is passed in the main `dehl` registers.

For _math48_ the fastcall registers are either the alternate register set or `FA` so there's no need to jump through hoops to preserve dehl.

I think ultimately that we should treat `__z88dk_fastcall` the same way as we treat `__z88dk_callee`:

```
extern double __LIB__ sqrt(double)
extern double __LIB__ sqrt_fastcall(double) __z88dk_callee;
#define sqrt(a) sqrt_fastcall(a)
```

That is a regular function call gets  the `_fastcall` variant. A function pointer call gets the non-fastcall variant.

This is leading to a separate header file for each maths library (which is probably fair enough anyway since the limits will be different), some m4 magic and restoring of those fastcall stubs.

feilipu, 2019-09-19:
The sccz80 [fn_fastcall stubs have been returned](https://github.com/z88dk/z88dk/commit/393b52db0e74f00cc9e599acf958523f2aed8ca8).
And the limits for [IEEE 754 are in this header file](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math32/c/m32_math.h). Out of this, I guess the individual library headers can be generated.

feilipu, 2019-12-08:
Just the newlib scanf core holding this open now.

feilipu, 2020-01-13:
Add the missing `float fmax (float,float)` and `float fmin (float,float)`, and removed some irrelevant functions.

@suborb I'm confused again whether the calling convention for sccz80 is stack left, register right for some functions (the intrinsics like `l_f32_mul`) but for other `__callee` functions it is stack left, stack right. That's correct I think, but is that the whole story for two variable operators?

Also I think that my whole linkage for `<math.h>` functions is fastcall functions is wrong. There is no `function_fastcall(float)` in sccz80. The function is purely called `function(float)` with a fastcall decorator, right?

There are quite a few `function_fastcall()` functions in the lm32 directory, with linkage for sdcc classic. I guess that these are only useful for that purpose, but aren't used for sccz80 at all.

In any case, I think I have to go back and fix all of the functions that are "intrinsic" and take a look at the C sourced functions again, because they move their parameter from stack to register, but this is going to overwrite the correct parameter already in the register.

Seems I still need to fix some stuff.

suborb, 2020-01-13:
You've got the calling convention right, the frame looks like this for `__smallc` functions:

    left hand parameter
    right hand paramter
    return address

The `_fastcall()` functions should be able to be shared between sdcc and sccz80 assuming there's the appropriate macro magic in `<math.h>` - there's only one parameter and it's in registers so it's exactly the same.

feilipu, 2020-01-13:
I think the problem that I have to unwind is that all the sccz80 fastcall functions are called as `function`, but in sdcc they are called as `_function_fastcall`. So that needs to be fixed across the board.

Also, if I'm not mistaken, intrinsic functions, like `l_f32_mul` are called with the right hand parameter in registers. At least that's what is set up now. Is the difference based on inclusion in `<math.h>` or not?

feilipu, 2020-01-14:
With the exception of scanf, which is currently unsupported for floats, this is done.
A new issue can be raised for scanf floats.

I'll close it now.

closed 2020-01-14 by feilipu