z88dk/z88dk #1156: (math32) fsadd/fssub issue

suborb, 2019-05-21:
Just getting the maths framework tests running for math32. And ran into an interesting issue in that the -- tests weren't working. 

Investigating further, the following have a result of 0:

```
3.0 - 1.0
3.9 - 1.0
3.9 - 1.9
```

And values in-between the ranges on both sides.

feilipu, 2019-05-21:
I've just run a number of tests with sdcc.newlib.rc2014.cpm, and I'm not seeing the issue.
A number of add/subtract scenarios all work, similar to this bug.
Will have a further look tonight.

suborb, 2019-05-21:
Try the following:

```
#include <stdio.h>

float func() {
#asm
        EXTERN  m32_fsadd_callee
        ld      de,0x4040               ;3
        ld      hl,0x0000
        push    de
        push    hl
        ld      de,0xbf80               ;-1
        ld      hl,0x0000
        call    m32_fsadd_callee
        ret
#endasm
}

int main() {
        float f;
        unsigned long l;
        f = func();
        l = *(unsigned long *)&f;

        printf("%f %08lx\n",f,l);
}
```

Compile with `zcc +test main.c -lmath32 -fp-mode=ieee` and run with `z88dk-ticks a.bin`

feilipu, 2019-05-22:
Yes... was a bit overzealous in shaving ticks.
Created a case where no shift would result in no carry, and exit to zero.

commit da51d186 (2019-05-22): m32 - fix #1156
closed 2019-05-22 by feilipu