z88dk/z88dk #1170: Building with newlib.

jayacotton, 2019-06-01:
Compiling this program is producing the errors listed below.  The +cpm switch does
not seem to be honored, a similar bug in stdlib.h the wchar_t barrier is not handled
correctly.  Its a typedef .... and defined in stddefs.h

#include "/home/jay/z88dk/include/_DEVELOPMENT/sccz80/stdio.h" 
#include "/home/jay/z88dk/include/_DEVELOPMENT/sccz80/string.h" 
#include "/home/jay/z88dk/include/_DEVELOPMENT/sccz80/stdlib.h" 

FILE *in;
FILE *out;
FILE *tmp;
char infilename[80];
char outfilename[80];
char tmpfilename[80];
char c;
char *p;
char buffer[1024];
int main(int argc, char *argv[])
{
	memset(infilename,0,80);
	memset(outfilename,0,80);
	memset(tmpfilename,0,80);
	strcpy(infilename,argv[1]);
	strcpy(outfilename,argv[1]);
	strcpy(tmpfilename,"C:XYZZY.TMP");

	in=fopen(infilename,"r");
	tmp = fopen(tmpfilename,"rw");
	if(tmp == 0){
		perror("can open tmpfile");
		exit(0);
	}
	while(p=fgets(buffer,1024,in)){
		strupr(buffer);
		fputs(buffer,tmp);
	}
	fclose(in);
	fflush(tmp);
	fclose(tmp);
}

This is the Makefile
toupper: main.c
	zcc +cpm -O3 -vn -s -m --list -clib=new -c main.c
	zcc +cpm -create-app -clib=new -otoupper main.o

clean:
	rm *.o *.err *.lis *.def *.sym *.exe *.COM macro

install:
	sudo cp ./*.COM /var/www/html/.

Here are the errors.

 make
zcc +cpm -O3 -vn -s -m --list -clib=new -c main.c
zcc +cpm -create-app -clib=new -otoupper main.o
Error at file 'fcntl/z80/asm_vopen.asm' line 60: symbol 'asm_target_open_p1' not defined
Error at file 'fcntl/z80/asm_vopen.asm' line 127: symbol 'asm_target_open_p2' not defined
Errors in source file /home/jay/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/cpm/cpm_crt.asm:
Error at file 'fcntl/z80/asm_vopen.asm' line 60: symbol 'asm_target_open_p1' not defined
Error at file 'fcntl/z80/asm_vopen.asm' line 127: symbol 'asm_target_open_p2' not defined
make: *** [Makefile:3: toupper] Error 1

suborb, 2019-06-01:
Unfortunately newlib doesn't support fopen etc for files for any target.

If you need to use files, then you'll have to use classic.

jayacotton, 2019-06-01:
well then,closed

closed 2019-06-01 by jayacotton

feilipu, 2019-06-01:
Jay, if you’re looking for a CP/M based solution, I’ve found the Hi-Tech C compiler works fine with standard file management. files. It’s clunky, and weird to use. But it is an option.
P.